lumapu/ahoy #1580: Mit ePaper kein nrf24 Empfang

Loetnase, 2024-04-09:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.106

### Github Hash

435aa17

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Mit ePaper habe ich keinen nrf24 Empfang.
ePaper deaktiviert, funktioniert die Kiste
[image]
[image]

### Error description

Mit ePaper habe ich keinen nrf24 Empfang.
ePaper deaktiviert, funktioniert die Kiste
![NRF24 mit Epaper](https://github.com/lumapu/ahoy/assets/120967249/daff8a19-aa1f-471
[image]
2-9e2c-032712fc3e6c)

lumapu, 2024-04-09:
sehr gerne 😊

knickohr, 2024-04-10:
Einspruch euer Ehren !

Das Problem tritt erstmalig mit der Version 0.8.104 auf. Alle Versionen davor funktionieren mit ePaper und nRF am ESP32.

Ich muß wohl wieder öfters offizielle Ahoy-Versionen testen anstatt mich in ZE verewigen 🤪

edw00d4711, 2024-04-10:
Bei mir lief es auch vorher. Jetzt leider nicht mehr. Und ich komme von weit unten. 231230_ahoy_0.8.35_86ff9f0_esp32.bin und davor lief es auch. Ich musste mal eines Tages das Verzeichnis einmal aufräumen :) Bitte wieder fixen.

Vielen Dank!

knickohr, 2024-04-10:
103 funktioniert

Loetnase, 2024-04-12:
@MetaChuh @knickohr 
zumindest keine positiven, ich habe heute Morgen mal die 0.8.110 eingespielt, das brachte keine Änderung. 
Dann schrittweise zurück, 0.8.103 läuft wieder, ab 0.8.104 geht mit ePaper nichts mehr.
In der Web Konsole bleibt das nach NTP bei der 104 einfach stehen.
[image]
Mit deaktiviertem ePaper sieht das dann so aus, da kommen die Radio Infos und los geht es.
[image]
Mit der 0.8.110 sieht das genau gleich aus.
Der Aufbau läuft schon längere Zeit, normaler ESP32, DukeWolle Board, Ebyte, ePaper 
Kannst ja mal testen aber oben schrieb lumapu das das aktuell nicht gehen kann.
Wäre aber schön wenn du was findest

knickohr, 2024-04-19:
@lumapu 

Ist das mit dem Fusion-Board gegengeprüft ?  Es sind mittlerweile mehrere Hundert Boards im Umlauf, da kann man nicht so schnell mal ein paar Pins verändern 😲

Das würde mächtiges Geschrei geben 🫣

lumapu, 2024-04-19:
@knickohr genau das ist mir heute Nacht auch gekommen - ich hatte ja auch geschrieben, dass noch keine Tests stattgefunden haben (oder nur sehr rudimentäre). Für das Fusion Board ist definitiv noch eine Änderung nötig. Danke für den Gedankenanstoß.

Loetnase, 2024-04-19:
@lumapu @knickohr 
Das sieht für mich gut aus, 
die Standart 0.8.111 vom Server zeigt bei mir noch immer den gleichen Fehler wie oben,
die Zeit wird geholt und das war es, mit der 0.8.103 funktionier es aber.
Mit Display deaktiviert an Standart 0.8.111 funktioniert es auch wieder, also identisch zu meinem Post von oben. 

[image]

aber mit dem spi_fix von oben funktioniert es bei mir am normalen ESP32, DukeWolle Board, Ebyte, ePaper,
ruckzuck kamen nach dem Neustart die Werte
[image]

Ich danke dir bzw. euch für den unermüdlichen Einstz

Edibo63, 2024-04-19:
Habe von 0.8.103de auf preview 0.8.111 upgedatet.
NRF streht wieder auf connected.

Loetnase, 2024-04-22:
@lumapu @knickohr 
Das sieht für mich gut aus, 
die Standart 0.8.111 vom Server zeigt bei mir noch immer den gleichen Fehler wie oben,
die Zeit wird geholt und das war es, mit der 0.8.103 funktionier es aber.
Mit Display deaktiviert an Standart 0.8.111 funktioniert es auch wieder, also identisch zu meinem Post von oben. 

[image]

aber mit dem spi_fix von oben funktioniert es bei mir am normalen ESP32, DukeWolle Board, Ebyte, ePaper,
ruckzuck kamen nach dem Neustart die Werte
[image]

Ich danke dir bzw. euch für den unermüdlichen Einstz
Das hatte ich vor Tagen schon geschrieben, aber wohl nicht abgeschickt
Aber jetzt verspätet :-)

lumapu, 2024-04-22:
jetzt muss ich es nur noch für das Fusionboard oder den ESP-S3 allgemein lösen

lumapu, 2024-04-22:
die Herausforderung wäre ESP32-S3 (aka Fusion) mit ETH, ePaper, NRF und CMT 😉. Ich hoffe die 0.8.112, die jetzt dann veröffentlicht wird kann das.

edw00d4711, 2024-04-23:
Moin zusammen,

danke für die kontinuierliche Weiterentwicklung. 
Leider hab ich nicht alles verstanden... :) Was muss ich jetzt wie stecken und im Menu konfigurieren? Gibt es dafür ein Fritzing?

Danke!

Loetnase, 2024-04-23:
@edw00d4711 da brauchst du aber mindestens einen ESP32 wie im Plan
CMT2300 anstatt NRF24 geht natürlich auch.
Wenn du mehr zum epaper wissen möchtest, komm rüber ins Discord #support-ahoy
[image]
[image]

edw00d4711, 2024-04-23:
@Loetnase Vielen Dank! Ich kontrollier erst mal meinen Aufbau und korrigiere den ggf. Ansonsten komme ich gerne in den Discord.

Edibo63, 2024-04-25:
0.8.112 läuft problemlos.
Vielen Dank.

Loetnase, 2024-05-05:
Also ich würde jetzt sagen das ursprüngliche Problem ist gelöst.
ePaper und NRF24 am ESP32 laufen bei mir wieder.

Dann hat sich bei mir aber leider das "Ghosting" am ePaper gezeigt welches ich vorher noch nie hatte.
Das werde ich aber noch ein bisschen anschauen und in einem neuen Issue melden. 
Schaut mal bei euch ob ihr da was am ESP32 und ePaper etwas vergleichbares seht.
[image]
Somit werde ich dieses Issue schliesen.
Danke für eure Mithilfe.

closed 2024-05-05 by Loetnase