lumapu/ahoy #1582: Retain Flag Homeassistant

slg60, 2024-04-10:
Bin gerade dabei AHOY v 0.8.106 via MQTT in Homeassistant einzubinden. Das klappt soweit, bis auf TotalYieldDay und TotalYieldTotal. 
Die Sensoren ändern ihre Werte nur beim Neustart und auch nur das eine mal.
Nach meiner Recherche hat das mit dem Retain Flag zu tun was bei den Topics ja gesetzt ist. 
Da kommt HA wohl nicht mit klar. Das lässt sich aber auch nicht konfigurieren in der AHOY.

Ist das bekannt?
Gibt es Lösungen dafür?

knickohr, 2024-04-10:
Das mußte ja kommen !  Ihr mit eurem HA. Vielleicht sollte HA erst mal so programmiert werden das es den Sinn und Zweck von MQTT versteht. MQTT ist ein Push-Sevice bei dem neue Werte erst ausgegeben werden wenn sie sich auch ändern, nicht vorher und auch nicht regelmäßig.

Sorry, kein persönlich Angriff an Dich, slg60, aber in der Community kommen fast täglich Probleme mit HA und vor allem wie man MQTT dazu vergewaltigtes, alle x Sekunden alte Werte wieder neu aufzuwärmen.

Tschuldidung, mußte das mal los werden 😇

slg60, 2024-04-10:
Naja, die Werte für YieldDay und Total ändern sich andauernd, aber in HA werden sie nicht aktualisiert, vermutlich aufgrund des retain flags. Im Shelly 3em kann man das retain z.b deaktivieren, in Ahoy nicht. Werte wo das Retain nicht gesetzt ist kommen immer sauber bei Änderung rein im HA.

knickohr, 2024-04-10:
Dann würde ich das Problem mal im HA anpacken 😉

slg60, 2024-04-10:
Ich frage mich nur, warum ich in allen mir bekannten MQTT fähigen Geräten das retain abschalten kann? Scheint ja schon seine Berechtigung zu haben.

slg60, 2024-04-10:
Der Wert ändert sich bei 3,2kwp bei einem Tag wie heute quasi andauernd, dennoch kommen die Werte nicht im HA an! Man kann das ja im MQTT Explorer ganz einfach gegenchecken.

knickohr, 2024-04-10:
Vielleicht hilft das ?

https://github.com/tbnobody/OpenDTU/discussions/1849

slg60, 2024-04-10:
Ne da war ich auch schon drauf gestoßen....

slg60, 2024-04-11:
Das liegt am Retain Flag.... Habe mir jetzt nochmal händisch einen MQTT Sensor in der Yaml erzeugt. Wenn ich händisch werte da hin schicke, dann kommen die sofort im HA an. Retain flag nicht gesetzt. Heute morgen hat die DTU wieder angefangen daten zu senden. Werte kommen im HA nicht an. Retain flag gesetzt. Ob das nun ein Bug im HA ist weiß ich nicht, es würde sich aber umgehen lassen wenn man retain in der DTU deaktivieren könnte...

lumapu, 2024-04-11:
wenn würde aber das Retain-Flag global abgeschaltet werden, kann ich einbauen, ist nur ein boolean

knickohr, 2024-04-11:
Das soll aber bitte jetzt nicht heißen das dann ALLE Topics entweder retained oder nicht sind 😵

slg60, 2024-04-11:
Leute, ihr kennt euch sicherlich besser aus als ich und ich möchte nicht, dass jetzt etwas eingebaut wird was keinen Sinn ergibt. Ich sehe es nur konfiguriebar im shelly 3em. Des Weiteren bin ich mir nicht sicher ob das wirklich das Problem ist. Das Problem kann auch an CCU-Jack liegen.

lumapu, 2024-04-11:
der neue enable / disable schaltet das retain flag nur komplett aus, wenn es gesetzt ist, aber schaltet es nie zusätzlich ein.

slg60, 2024-04-11:
Ich habe jetzt mal die Ahoy direkt an den internen Mosquitto Broker angebunden. Mal sehen ob sich das auch fehlerhaft verhält. Morgen früh weiß ich mehr.

lumapu, 2024-04-11:
ok, bleibt spannend

slg60, 2024-04-12:
Das Problem scheint im CCU Jack angesiedelt zu sein. Mit Mosquitto als Broker läuft alles wie gewünscht.
Ich denke es ist dennoch gut das man das retained jetzt konfigurieren kann. Issue im Projekt CCU-Jack ist platziert. Scheine mit dem Problem auch nicht alleine zu sein...

cgatnet, 2024-08-30:
Ergänzend: Info: Heads Up:
Man sollte beachten, dass, wenn man damit rumspielt (is ja ein Häkchen, was macht man damit? :o) und es "abhakt" (disabled), dann bleibt der letzte Wert der "Retained Topics" in Mosquitto (oder anderen Brokern) erhalten. Dies kann zu rätselhaften Phänomenen führen... Daher: immer schön gesetzt lassen, wenn man sich damit nicht so gut auskennt! ;o)

stefan123t, 2024-11-02:
@cgatnet sollen wir einen modalen Dialog dazwischen / danach schalten ? 
Wenn man es deaktivieren will, kommt dann eine Warnung: "Willst Du das wirklich deaktivieren, das ist potentiell gefährlich. Ja / Nein".

cgatnet, 2024-11-03:
> @cgatnet sollen wir einen modalen Dialog dazwischen / danach schalten ? Wenn man es deaktivieren will, kommt dann eine Warnung: "Willst Du das wirklich deaktivieren, das ist potentiell gefährlich. Ja / Nein".

Entweder das, oder: evtl. reicht es schon die Warnung in die Hilfe aufzunehmen oder „neben das Häkchen“ zu schreiben? 🤔