lumapu/ahoy #1583: [Q] Versionsstring für Prometheus über die API?

fra-wwilhelm, 2024-04-10:
## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [ ] Raspberry Pi

### Version / Git SHA: 
Version: 240217_ahoy_0.8.83_5ebfe5a_esp32-wroom32-prometheus.bin

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [X ] ESP Tools

Hallo,

ich versuche die Daten direkt mit Grafana zu visualisieren, dort erhalte ich aber eine Fehlermeldung beim Einrichten der Prometheus Datenquelle:
[code]
Post "http://192.168.178.240/api/v1/query": EOF - There was an error returned querying the Prometheus API.
[/code]
IP der AhoyDTU ist 192.168.178.240
per curl erhalte ich auch Werte:
[code]
~$ curl http://192.168.178.240/metrics
# TYPE ahoy_solar_info gauge
ahoy_solar_info{version="0.8.83",image="",devicename="AHOY-DTU"} 1
...
[/code]
und die API kennt den Einstiegspunkt ja auch nicht:
[code]
http://192.168.178.240/api
{
    "avail_endpoints": {
        "inverter/list": "http://192.168.178.240/api/inverter/list",
        "inverter/id/0": "http://192.168.178.240/api/inverter/id/0",
        "inverter/alarm/0": "http://192.168.178.240/api/inverter/alarm/0",
        "inverter/version/0": "http://192.168.178.240/api/inverter/version/0",
        "generic": "http://192.168.178.240/api/generic",
        "index": "http://192.168.178.240/api/index",
        "setup": "http://192.168.178.240/api/setup",
        "setup/networks": "http://192.168.178.240/api/setup/networks",
        "setup/getip": "http://192.168.178.240/api/setup/getip",
        "system": "http://192.168.178.240/api/system",
        "live": "http://192.168.178.240/api/live",
        "powerHistory": "http://192.168.178.240/api/powerHistory"
    }
}
[/code]
Wo muss ich jetzt ansetzen das Grafana als Datenquelle diese DTU annimmt?

LG

lumapu, 2024-04-11:
@fsck-block darf ich wieder um deine Hilfe bitten? Ich selbst betreibe kein Prometheus und müsste mich erst einarbeiten.

fsck-block, 2024-04-12:
Hallo @fra-wwilhelm 

> Wo muss ich jetzt ansetzen das Grafana via Prometheus als Datenquelle diese DTU annimmt?

Das Ganze ist mehrstufig. 
1. Prometheus sammelt die Daten von der DTU und speichert sie in einer speziellen Datenbank
2. Grafana nutzt dann die Daten von Prometheus und visualisiert sie nach deinen Angaben.

Also fangen wir bei Prometheus an. Um Grafana kommt erst in's Spiel wenn Prometheus läuft

Prometheus nutzt nur den Link auf die `http://192.168.178.240/metrics` und geht nicht über `/api`. 
Du musst in deiner Promehtheus Konfiguration `prometheus.yml` einen Job für deine DTU eintragen. 
Das könnte in etwa so aussehen
``` 
 - job_name: 'ahoydtu'
    static_configs:
      - targets: ['192.168.178.240:80'] 
        labels:
            instance: ahoy-dtu.local
```
Nach einem Neustart von prometheus kannst du dann die Daten erst mal über die Weboberfläche von prometheus prüfen.
Bei mir sieht das z.b. so aus
[image]

Bitte gib' Bescheid ob's soweit geklappt hat.

fra-wwilhelm, 2024-04-12:
stupid me,

ich bin fälcshlichreweise davon ausgegangen das AhoyDTU _auch_ einen Prometheus servre (DB) implementiert, macht er natürlich auch nicht. die AhoyDTU FW stellt ja nur die Metriken bereit.
Also: 
````
sudo mkdir -p /volumes1/docker/prometheus
sudo chown 65534.65534 /volumes1/docker/prometheus
sudo bash -c 'cat <<EOF >  /volumes1/docker/prometheus/prometheus.yml
global:
 scrape_interval: "30s"
 evaluation_interval: 5s
 external_labels:
  env: "Development"

scrape_configs:
 - job_name: "ahoydtu"
   scheme: http
   metrics_path: "/metrics"
   tls_config:
    insecure_skip_verify: true
   static_configs:
   - targets: ["192.168.178.240"]
     labels:
      site: garage
EOF'

docker run \
    -p 9090:9090 \
    --name prometheus
    -v /volumes1/docker/prometheus/prometheus.yml:/etc/prometheus/prometheus.yml \
    -v /volumes1/docker/prometheus/data:/prometheus \
    --restart-always \
    prom/prometheus
EOF'

In Grafana dann natürlich den Prometheus Server eintragen http://127.0.0.1:9090

fra-wwilhelm, 2024-04-12:
Vielen Dank für die schnelle Hilfe, die Anfrage kann geschlossen werden.

closed 2024-04-12 by fra-wwilhelm