lumapu/ahoy #1585: [Bug] opendtufusion CMT2300A keine Connection zu Invertern nach Stromausfall/Unterbrechung Stromversorgung

tester277, 2024-04-12:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.110

### Github Hash

0.8.110

### Build & Flash Method

was already installed

### Setup

es ist nur CMT2300A aktiv, nrf ist inaktiv

### Error description

Erstmal vielen Dank für die tolle Arbeit!
Ich habe ein komplettes Opendtufusion Gerät im Einsatz.
Nun hatte ich einen Stromausfall.
Nach dem hoch fahren ist das Gerät Netzwerk seitig erreichbar, allerdings wird die Verbindung zu den Invertern nicht mehr neu hergestellt. Unter Systen wird Interrupt Pin working false angezeigt.
Mein workaround sieht so aus, dass ich alle (3) Inverter entferne und nach Neustart erstmal einen Inverter konfiguriere. Dann ist Interrupt Pin working wieder True. Danach kann ich mein config Backup einspielen und alles funktioniert wieder. Nur Konfig einspielen oder reboot behebt das Problem nicht, ebenso nicht wenn ich ein paar Versionen zurück gehe.
Kann dieses Problem mit der openfusion dtu bestätigen?
Was wäre noch hilfreich?
Vielen Dank

tester277, 2024-04-12:
@MetaChuh 
Nach Rückupdate auf 0.8.91 tritt der Fehler erwartungsgemäß nicht auf.
Nach Unterbrechung der Versorgung schon.
19:04:11.567 -----
19:04:11.568 W: CMT TX failed, code: 2
19:04:11.804 W: CMT TX failed, code: 2
19:04:11.804 -----
19:04:11.805 W: CMT TX failed, code: 2
19:04:11.941 W: CMT TX failed, code: 2
19:04:11.941 -----
19:04:11.942 W: CMT TX failed, code: 2
19:04:12.095 W: CMT TX failed, code: 2

tester277, 2024-04-26:
Leider tritt das Problem nach Unterbrechung der Spannungszufuhr (auch 0.8.112) weiterhin auf.
Habe einen optimierten Workaround mit dem ich das Problem (glücklicherweise auch Remote) beheben kann.
Nach Spannungsausfall > erase Settings (außer Wifi) > in pinout config nRF24L01+ deaktivieren und CMT2300 aktivieren > Kontrolle nach reboot unter System CMT steht auf unknown (gutes Zeichen) > Settings importieren > läuft
Für mich hat es den Anschein als würde der CMT einen Init nach Unterbrechung benötigen.
Dagegen spricht , dass ich wohl der Einzige bin bei dem das auftritt.

lumapu, 2024-04-29:
warum genau die 0.8.91 @MetaChuh ?

Hans80939, 2024-05-13:
Bei mir funktioniert der Interrupt-Pin auch nicht, aber eigentlich nie:
12:38:40.137 -----
12:38:40.137 W: CMT TX failed, code: 2
12:38:40.373 -----
12:38:40.373 I: com loop duration: 373ms
12:38:40.373 -----
12:38:55.137 -----
12:38:55.137 W: CMT TX failed, code: 2
12:38:55.373 -----
12:38:55.373 I: com loop duration: 373ms
12:38:55.373 -----

Der Programmstand 0.8.91 funktioniert bei mir nicht, weil ich eine Seriennummer HMS-800-2t habe, die auch Buchstaben enthält. Die Eingabe von Buchstaben funktioniert bei mir mit 0.8.118. Aber dort wirde der Interrupt-Pin nicht erkannt.

Meine Anschlusseinstellungen:
CMT2300A Funkmodul aktivieren
SCLK GPIO6
SDIO GPIO5
CSB GPIO4
FCSB GPIO21
GPIO3 GPIO8

tester277, 2024-05-13:
@Hans80939 und somit hast Du mein Problem gelöst!
Vielen Dank
Es war (doch) die Konfiguration als Ursache.
Statt FCSB GPIO21 war FCSB GPIO1 (warum auch immer).
Nach Unterbrechung der Spannungsversorgung ist mit GPIO21 auch kein Problem mehr vorhanden.
Danke auch an die anderen fleißigen Helfer!

tester277, 2024-05-13:
Pinout Konfiguration war falsch 
Statt FCSB GPIO21 war FCSB GPIO1 konfiguriert

closed 2024-05-13 by tester277

Falerion-git, 2024-08-11:
Ich habe ein ähnliches Problem, dass der Interrupt als  "false" angezeigt wird (opendtufusion board).

Mir war aufgefallen, dass mit den dev Versionen 8.127, 8.130 oder 8.134  (andere nicht ausprobiert) der Interrupt Pin als "false" angezeigt wird. Wenn ich auf die Version 8.83 zurück kehre, dann bleibt der Interrupt Pin zunächst auf "false".
Erst wenn ich in den Einstellungen des Inverters unter "Radio" die "Frequency" auf 866.00 Mhz setze bekomme ich den Inverter direkt wieder angezeigt. Nach dem downgrade von den dev Versionen ist die zunächst auf 860.00 MHz eingestellt.
Vor dem Upgrade hatte ich auch die Frequenz von 866.00 MHz genutzt.

Der Fehler code war wie im issue oben:  CMT TX failed, code: 2
Nach dem Frequenzwechsel ist dieser Fehler auch verschwunden.

Ich bin ursprünglich auf eine der dev Version gewechselt, weil ich jetzt einen zweiten Inverter einbinden wollte, der eine Seriennummer mit Buchstaben hat. Leider schient es die Option, die Frequenz einzustellen nicht mehr zu geben (zu mindestens in den ausprobierten dev Versionen (siehe oben)). Der ursprüngliche Inverter wird mir in der Übersicht als "enabled" angezeigt, aber er liefert keine Daten. Wenn ich die "Active Power Control" aufrufe und dann einen "Restart" des Inverters anstoße, bekommen ich eine "Error: authentification failed" Fehler. Vermutlich aufgrund des nicht funktionierenden Interrupts.

Wird die Frequenz jetzt eigentlich automatisch eingestellt?
Kann ich in den dev-Versionen irgendwie die Frequenz manuell einstellen?

unabhängig vom obigen Problem:
Bei den dev Versionen ist mir aufgefallen, dass wenn ich einen zweiten Inverter mit Buchstaben Seriennummer hinzufüge, ich den nicht aktivieren kann. (Erster Inverter ohne Buchstaben in der Seriennummer). Ein setzen der "Enabled" Box wird beim speichern ignoriert.
Zudem wird immer nur ein "Input" (also die Panel Eingänge) als konfigurierbar angezeigt. Weitere "Input" Felder lassen sich nicht hinzufügen, bzw. sind nicht vorhanden.