lumapu/ahoy #1589: Nach WLan Nachschaltung ist die DTU morgens immer im AP Modus

Ollipop030, 2024-04-16:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.110

### Github Hash

4587c1c

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

"Reboot at midnight" is off.
"Start without time sync" is off.

### Error description

Bei mir wird Nachts das WLan an der Fritzbox abgestellt, morgen um 7 wieder gestartet. Mit der Version 0.8.110 ist die DTU aber morgens im AP Modus, auch nach einer halben Stunde noch.
Das gleiche habe ich auch schon am Wochenende bemerkt, da war es die Version 0.8.108. Seit welcher Version das so ist kann ich nicht genauer sagen, ich kam direkt von der 0.8.97, die hat das Problem aber nicht. 
Oder habe ich irgendwelche Settings verpasst?

Gubi2023, 2024-04-16:
kann ich so nur bestätigen. siehe auch #1575

Ollipop030, 2024-04-16:
Ah, gar nicht gesehen.

Jawada74, 2024-04-16:
Kann ich auch bestätigen alles nach der 08.97 hatte das Problem bei mir.   Bin jetzt wieder auf der 08.97 die hat das Problem nicht

closed 2024-08-16 by lumapu