lumapu/ahoy #1591: MQTT send Button schick ungültige Informationen

rmayergfx, 2024-04-16:
extra hier nochmal mit einem neu aufgesetzten dev 0.8.110 getestet:
[image]

Im Hintergrund läuft zur Überwachung der MQTT Explorer mit. Sobald ich in den Settings von AhoyDTU auf den Button
Discovery Config drücke, werden die Topics an HomeAssistant geschickt, im MQTT Explorer erscheint dann sofort ch0_ALARM_MES_ID das aber ungültig ist:

homeassistant/sensor/xxxx/ch0_ALARM_MES_ID

Wie kann das sein?

knickohr, 2024-04-17:
Der Topic ist eigentlich obsolet. Die Daten kommen jetzt über einen JSON-Oayload unter dem Topic …/alarm_data

rmayergfx, 2024-04-17:
@knickohr 
Ich habe im Hintergrund MQTT Eplorer offen um zu sehen was passiert. Gehe ich in AhoyDTU auf den Button Send, erscheint automatisch ch0_Alarm_MES_ID, und das dürfte doch eigentlich gar nicht sein. Ich lösche das Topic und sobald ich auf den Button senden (Discovery Config ) gehe erscheint das Topic und das dürfte doch gar nicht sein, das macht mich stutzig.

knickohr, 2024-04-17:
Dann isr dieses Discovery nicht mehr in Ordnung.

rmayergfx, 2024-04-17:
Was meinst du damit? Fehler bei AhoyDTU?

lumapu, 2024-04-17:
ja, wird entfernt, danke für den Hinweis

rmayergfx, 2024-04-18:
Gerade getestet.... Fehler ist nicht mehr da....

knickohr, 2024-04-18:
Bei der Gelegenheit kannste LastAlarmCode auch gleich raus nehmen 😉