lumapu/ahoy #1592: calculation error

Peertje-62, 2024-04-17:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.11

### Github Hash

4587c1c

### Build & Flash Method

ESP Tools (flash)

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Hello, hope I can clarify a calculation error,
have 2 inverters; HM 600 Max 456 Watt at 3:54 PM, HM300 Max 306 Watt 11:19 AM, in the web gui (Total) this adds up to 762 Watt Max which is not correct. In ''History'' the Max is 541 Watt which is the correct value (Also according to Grafana) Kind regards
[image]
[image]

hrolofs, 2024-04-17:
It looks that 762 are the sum of both max values and 541 are the max of the summary value of both inverter.

kiu77, 2024-04-17:
Yes, this is not the first time mentioned. We have to wait until this will be fixed someday ;-)

lumapu, 2024-04-17:
I'm waiting too 😂 - just a joke - have that on my list but was not solved until now

Peertje-62, 2024-04-18:
Good Morning,
Must have missed it
Thx for the answer anyway
Have a nice day
regards ;-)

closed 2024-04-18 by Peertje-62