lumapu/ahoy #1593: 0.8.111 sehr instabil auf neuen DTUs

kiu77, 2024-04-18:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.11

### Github Hash

123

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

ESP32 mit nRF24 und CMT2300A

### Debug Serial Log output

```shell
no
```

### Error description

Während auf "meiner" DTU mit nRF24L01+ und CMT2300A an einem HM-800 und einem HMS-800
alles auch unter 0.8.111 prima läuft, ist das Verhalten mit zwei unterschiedlichen Board mit 30-pol. und
38-pol. ESP32 komplett erratisch, wenn man von 0.8.83 auf 0.8.111 updated.
Ohne bzw. mit den Default-Settings kommt man immerhin in den AP-Mode.
Stellt man schon unter 0.8.8.3 das Display + Wechselrichter ein, klappt das auch bei den beiden
jungfräulichen Testboards. Nach anschließendem Update auf 0.8.111 hingegen gibt es häufige
Reboots auch bei Klicks auf die Weboberfläche und die Wechselrichter werden nicht angesprochen,
Die Interrupt-Pins bleiben unter "System" gelb.
Gleiches Verhalten wenn ich direkt nach neuem Flashen auf 0.8.3 mit Erase der Settings zuerst
auf 0.8.111 update: Das Display kriegt man dann noch aktiviert, aber nach Eintrag von einem
oder auch beiden Wechselrichtern kein Kontakt zu diesen.
Der einzige Unterschied zu "meiner" funktionierenden DTU ist, dass diese langsam über etliche
Versionen bei 0.8.111 angelangt ist.
Wenn ich die Einstellungen von "meiner" DTU ex- und in den neuen importiere, und dann statt dem
AP-Mode die korrekte WLAN-Adresse + PW eingebe, klappt alles.
Ist doch merkwürdig, oder?

[image]

kiu77, 2024-04-20:
Habe ich gemacht, aber es gibt keinen wirklichen unterschied im Verhalten.

Ich habe jetzt eine ungefähre Ahnung, wo es hängt:
Der AP-Modus macht ein Problem in 0.8.111
Beim Testen im AP-Modus stört mit der WiFi-Wizzard und ich beende den direkt, um dann per Klick auf "AhoyDTU" die Zeit direkt vom Browser zu holen. Die holt er zwar, aber es wird keine Abfrage der eingetragenen Inverter gestartet.
Das Gleiche ist auch der Fall, wenn ich bei Einstellungen -> Wechselrichter den Eintrag "Kommunikation starten ohne gültige Zeit (sinnvoll im AP Modus)" angehakt habe. Auch dann wird kein Inverter abgefragt.
Soweit ich das sehe, war das bisher anders und das Verhalten ist durchweg mit mehreren DTU-Aufbauten und unterschiedlicher Bestückung mit 30/38-pol. ESP32 und entweder nRF24 oder CMT2300A oder beiden reproduzierbar.
Das stört die Einrichtung zumindest bei mir.

0.8.111 funktioniert gut, wenn die DTU nicht im AP-Mode ist, also sich ins heimische WLAN einloggt und per DHCP eine IP-Adresse zugewiesen bekommt.

Weiter dauert es bei den höheren Versionen länger, bis die DTU nach Start (Strom anlegen) den AP-Mode aktiviert bzw. bis seine IP-Adresse im WiFi-Client meines Rechner sichtbar wird. Gelegentlich versagt das Einloggen dann mit dem PW ESP_8266 und die DTU rebootet.

Der WiFi-Wizzard zeigt zudem keine Auswahl der verfügbaren Netze an, funktioniert aber, wenn man die SSID (samt PW) manuell eingibt.

kiu77, 2024-04-21:
@lumapu 
@MetaChuh 
Sodele, damit die Konfiguration mit der neusten Version klar ist, habe ich ein fünfseitiges PDF mit Screenshots und einer Schritt-für-Schritt-Anleitung erstellt für alle, die das interessiert. Die Screenshots sind mit der deutschen Version der Firmware gemacht.

[AhoyDTU konfigurieren.pdf](https://github.com/lumapu/ahoy/files/15051610/AhoyDTU.konfigurieren.pdf)

Gubi2023, 2024-04-21:
Super! Endlich mal ne ausführliche Anleitung!

kiu77, 2024-04-22:
Zusammengefasst zur Version 0.8.111
1. Es werden keine Inverter im AP-Mode abgefragt = wenn sich die DTU nicht ins WLAN eingeloggt hat.
    Wenn im WLAN eingeloggt, funktioniert sie gut.
3. Bis das WLAN des AP-modes aufgebaut wird, dauert es sehr lang. Manchmal klappt es garnicht richtig.

lumapu, 2024-04-22:
der Wizard ist buggy, das weiß ich und seht auch noch auf dem Plan. Danke für deine Untersuchungen. Scheinbar habe ich in einer der letzten Versionen das Abfragen im AP-Modus zuerstört, auch das werde ich mir anschau
Sehr schöne Anleitung, auch von der Aufmachung, erinnert ein bisschen an eine Zeitung mit den 3 Spalten.

kiu77, 2024-04-22:
Wenn Du wills, kann ich für die nächste stable release eine Anleitung für alle machen, die Du verlinken kannst, falls sie Deinen Qualitätskriterien entspricht.

lumapu, 2024-05-30:
der Wizard wurde zwischenzeitlich schon überarbeitet, ich hoffe er ist jetzt besser zu bedienen. Leider blockiert ein Scan nach WLANs alles für ca. 5s. Ich weiß nicht, ob das normal ist oder nur bei meiner Implementierung so ist.
Bitte weiterhin berichten, damit wir evtl. hier noch besser werden.

lumapu, 2024-08-13:
gibt's hier einen neuen Stand?

stefan123t, 2024-11-02:
@kiu77 besteht das Problem auch beim Installieren / Upgrade auf 0.8.140 bzw. development03 0.8.152 noch ?

Bzw. umgekehrt können wir es einfach als gelöst markieren, weil das Upgrade auf 0.8.111 aktuell kaum mehr jemand macht und es bei späteren Versionen siehe oben vielleicht nicht mehr auftritt ?

kiu77, 2024-11-02:
Es tritt bei .140 ff. nicht mehr auf. Also kann man den Thread schließen.

closed 2024-11-02 by kiu77