lumapu/ahoy #1594: Feature Request :  Einführen eines weiteren Status „Standby“

knickohr, 2024-04-18:
Aufgrund von „Zero Export“ kommt es immer wieder vor das Inverter „soft“ ausgeschaltet werden und sich somit quasi im Status „Standby“ befinden ohne jedoch wirklich offline zu sein. Die Inverter können je nach Zeitpunkt wann sie Standby geschaltet werden dann in dem herkömmlichen Status 1 (available but not producing), Status 3 (available and was producing) oder sogar im Status 4 (was available) befinden. Das ist etwas verwirrend.

Vorschlag :  Einführen eines weiteren Inverterstatus 5 (Standby) wenn der Inverter von Zero Export oder auch von Hand „Turn off“ geschaltet wird.  Man kann diesen Status auch an der Alarm-ID 124 (Shut down by remote control) erkennen). Selbstverständlich soll wieder in den alten Status gewechselt werden wenn der Inverter den Staus Standby verläßt.

lumapu, 2024-04-22:
der ursprüngliche Status ist doch nach paar Minuten obsolet oder täusche ich mich da? Anhand den geschilderten Informationen könnte man sowas schon einbauen.

knickohr, 2024-04-22:
[image]

Der grüne Inverter wird von ZE immer wieder abgeschaltet (soft über „Turn off“), und es sieht in der State-Machine dann immer so aus als ob er im Status 1 ist. Kann aber je nach Dauer auch 4 oder gar 0 werden.

lumapu, 2024-04-22:
ok, danke für die Info der verzögerten Alarm-ID, da wäre ich voll reingetappt.