lumapu/ahoy #1595: v0.8.111 MQTT sendet Publish "Powerlimit" mit falschen Appl Daten, nach Topic (position 40+4)  kommt  '\0', '\x0E', '1', '0', '0',:. Es sollte nur "' '1', '0', '0'" für Powerlimit kommen

jotroj, 2024-04-20:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.111

### Github Hash

 AhoyDTU © 2024      Discord     Github      GIT SHA: 388cd89 :: 0.8.111

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

keine Änderungen zur vorherigen Vesion im Setup

### Debug Serial Log output

```shell
0.8.111 - 388cd89
---------------
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
webSerial, buffer overflow!
```

### Error description

MQTT - Datensendung : Fehler liegt bei 0,14
'<', '/', '\0', '(', 'i', 'n', 'v', 'e', 'r', 't', 'e', 'r', '/', 'F', 'l', 'a', 'c', 'h', 'd', 'a', 'c', 'h', '/', 'c', 'h', '0', '/', 'a', 'c', 't', 'i', 'v', 'e', '_', 'P', 'o', 'w', 'e', 'r', 'L', 'i', 'm', 'i', 't', '\0', '\x0E', '1', '0', '0'
60, 47, 0, 40, 105, 110, 118, 101, 114, 116, 101, 114, 47, 70, 108, 97, 99, 104, 100, 97, 99, 104, 47, 99, 104, 48, 47, 97, 99, 116, 105, 118, 101, 95, 80, 111, 119, 101, 114, 76, 105, 109, 105, 116, 0, 14, 49, 48, 48
0x3C, 0x2F, 0x0, 0x28, 0x69, 0x6E, 0x76, 0x65, 0x72, 0x74, 0x65, 0x72, 0x2F, 0x46, 0x6C, 0x61, 0x63, 0x68, 0x64, 0x61, 0x63, 0x68, 0x2F, 0x63, 0x68, 0x30, 0x2F, 0x61, 0x63, 0x74, 0x69, 0x76, 0x65, 0x5F, 0x50, 0x6F, 0x77, 0x65, 0x72, 0x4C, 0x69, 0x6D, 0x69, 0x74, 0x0,0xE, 0x31, 0x30, 0x30 
'<', '/', '\0', '(', 'i', 'n', 'v', 'e', 'r', 't', 'e', 'r', '/', 'F', 'l', 'a', 'c', 'h', 'd', 'a', 'c', 'h', '/', 'c', 'h', '0', '/', 'a', 'c', 't', 'i', 'v', 'e', '_', 'P', 'o', 'w', 'e', 'r', 'L', 'i', 'm', 'i', 't', '\0', '\x0E', '1', '0', '0'

lumapu, 2024-04-22:
die vorherige Version war 0.8.110?

lumapu, 2024-04-22:
woher kommt das "active PowerLimit", da sendet mWn. Ahoy nie?
Ich kann nicht vestehen was genau der Fehler ist und wie die erwartete Ausgabe ist. Es sind einfach zu viele Zahlen.
Btw. der Betreff ist keine Freitextfeld 😉

Airwalk007, 2024-04-25:
Irgendwas muss mit dem "active PowerLimit" aber nicht in Ordnung sein. Ich konnten den Fehler nachstellen (im ioBroker und mit HoymilesZeroExport-Skript):

So wird z.B. bei der Nutzung des HoymilesZeroExport-Skripts das Powerlimit geändet und dann von Ahoy als MQTT Nachricht verschickt.

Unglücklicherweise endet das mit einem "Received pubrel" und einem "Message xxxx deleted after 11 retries" ... hier mal ein exemplarischer Auszug aus dem Debug-Log für die MessageID 1642:

2024-04-25 14:34:04.060 - debug: mqtt.0 (20919) Client [ahoydtu] received publish package {"cmd":"publish","retain":false,"qos":2,"dup":false,"length":47,"topic":"inverter/HM-800-2/ch0/active_PowerLimit","payload":{"type":"Buffer","data":[50,55,46,54]},"messageId":1642}
.
.
.
2024-04-25 14:34:04.085 - debug: mqtt.0 (20919) Client [ahoydtu] received pubrel package {"cmd":"pubrel","retain":false,"qos":1,"dup":false,"length":2,"topic":null,"payload":null,"messageId":1642}
.
.
.
2024-04-25 14:34:29.917 - warn: mqtt.0 (20919) Client [ahoydtu] Message 1642 deleted after 11 retries

Komisch ist, das dieses Package mit qos:2 geschickt wird - während die meisten anderen packages (YieldDay, PC_AC, usw.) mit qos:0 gesendet werden.

Vielleicht hat jemand da eine Idee? Lässt man das HoymilesZeroExport länger laufen (und es finden viele Änderungen des Powerlimits statt) flutet ioBroker das Logfile mit "received pubrel" und "message xxxx deleted after 11 retries" ...

jotroj, 2024-04-29:
Habe den Fehler im eigenen Programm gefunden. Es war keine richtige Qos=2- Behandlung.

closed 2024-04-29 by jotroj