lumapu/ahoy #1596: Nach Crash verbindet sich Ahoy nicht mehr mit dem stärksten AP

knickohr, 2024-04-23:
Seit der Version 103 verbindet sich das WiFi von Ahoy nicht mehr mit dem stärksten verfügbaren AP nach einem ungewollten Crash.

Ein weiterer manueller Reboot löst das Problem wieder.

knickohr, 2024-04-23:
Nein, er verbindet sich bei mir mit dem zweitbesten AP

Gubi2023, 2024-04-23:
kann ich so bestätigen, das beste WLAN wird nicht mehr gefunden/angezeigt

knickohr, 2024-04-24:
Doch, angezeigt wird es. Und nach einem weiteren manuellen Reboor ist alles wieder OK.

lumapu, 2024-05-30:
wie ist hier der aktuelle Status? Hat sich was geändert, oder weiterhin vorhanden?

knickohr, 2024-06-06:
Leider ist es mit der 123 nachbeinem Crash immer noch 😢

knickohr, 2024-08-13:
Das Problem scheint neuerdings auf die neue Funktion „dynamisches Smart-Repeating“ geschuldet zu sein. Denn Ahoy verbindet sich jetzt immer erst mit dem stärksten AP, fällt aber nach einigen Minuten bis Stunden wohl auf den von der Fritz!Box zugewiesenen AP zurück.

closed 2024-08-13 by knickohr