lumapu/ahoy #1597: Wizard evven after configuration

Argafal, 2024-04-24:
0.8.111 [GIT SHA: 388cd89](https://github.com/lumapu/ahoy/commits/388cd89)

Wenn ich auf /visualization geht, öffnet sich /wizard. Mein AHOY ist schon konfiguriert und im WLAN. Klickt man auf "stop wizard", wird man auf die nicht existierende 192.168.4.1 umgeleitet. @Dahlienhaus konnte das Problem ebenfalls reproduzieren.

Argafal, 2024-04-24:
Is someone able to correct my typo in the title: "evven" -> "even"?

lumapu, 2024-04-24:
@Argafal das heißt nicht `/visualization` sondern `/live`. Aber allgemein ist da mit dem Wizard noch was im Argen

lumapu, 2024-04-25:
dad Ken man reproduzieren, indem man mit dem AP des ahoy verbunden ist, da dann der Default `http://192.168.4.1/wizard` ist. Wenn der AP aus ist, dann kommt man nach `http://192.168.4.1/index`

closed 2024-08-16 by lumapu