lumapu/ahoy #1599: DTU-PRO + AHYO-DTU

jacques3629, 2024-04-25:
## Hardware
  * [ ] ESP8266
  * [ok  ] ESP32
  * [ ] Raspberry Pi

Modelname: _____0.8.111_
Retailer URL: ______

### nRF24L01+ Module
* [ok ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Good morning

Sorry if this question has already been asked
I have an official dtu-pro gateway with 0 injection activated recently, I experience a lot of disconnection with the dtu-pro + ahyo-du
I imagine that access to the inverters cannot be shared?
when I unplug ahyo dtu no more disconnection with the dtu-pro gateway
does it have a setting to prevent this
I like ahyo-dtu I wish I could keep it, but I have to keep the 0 injection

Sincerely

jacques3629, 2024-04-26:
Thank you for your reply
the only solution is to support you so that the software evolves with the update of 0 injection in the near future of AHYO-DTU
to get rid of the official DTU-PRO gateway
The best of wishes for the future

Sincerely

lumapu, 2024-04-26:
we're currently in beta testing of our zero-export functionallity. So stay tuned by checking this repository again in approx. one month. There is also a discussion going on in Discord where you can join and maybe start as beta tester.

jacques3629, 2024-04-26:
Great news out of curiosity today, I tried the dev version this morning
AhoyDTU © 2024
Discord
Github
GIT SHA:b291ad1::0.8.112

not a single disconnection despite 0 injection
HM inverter
maybe lucky or not in the change log
#DevelopmentChanges

## 0.8.112 - 2024-04-24
* improved wizard
* converted ePaper and Ethernet to hal-SPI
* improved network connection

## 0.8.111 - 2024-04-17
* fix MqTT discovery field `ALARM_MES_ID` #1591
* fix Wifi reconnect for ESP32 #1589 #1575
* open link from `index.html` in new tab #1588 #1587
* merge PR: Disable upload and import buttons when no file is selected #1586 #1519

[image]

Sincerely