lumapu/ahoy #1600: [Bug] Viele Reboot mit ESP32 im laufe des Tages 0.8.113

blueline13, 2024-04-26:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.113

### Github Hash

177420f

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- ESP 32
- NRF
- CMT
- MQTT
- API
- KEIN DISPLAY
- Limit Regelung 

### Debug Serial Log output

```shell
Siehe Logfile am Ende
```

### Error description

Immer wieder Reboots während die Inverter aktiv sind.  Während der Nacht läuft die DTU durch. Das einzige was noch aufgefallen ist, wenn der Zeit Intervall in dem die DTU die Daten sendet. Steht er auf 10 Sekunden ist das deutlich weniger als wenn der Intervall auf 5 Sekunden steht. Ich habe mal die Teile der Reboot aus dem Serial log rausgeschnippelt und hoffe das es hilft. Wenn ihr noch etwas braucht was ich gerade vergessen habe, reiche ich es gerne nach.
[ESP32-Reboot.log](https://github.com/lumapu/ahoy/files/15134089/ESP32-Reboot.log)

SnakeGER, 2024-04-26:
Hi. Ich hatte das Problem bei früheren Versionen auch schonmal. Es lag bei mir ganz banal an der Stromversorgung. Ich hatte die DTU seinerzeit am USB-Anschluss meines Monitors angeschlossen. Nachdem ich der DTU ein eigenes Netzteil spendiert hatte (einfacher Samsung-Stecker fürs Handy) läuft alles problemlos. LG

blueline13, 2024-04-27:
@SnakeGER 
Danke für den Hinweis und ja das Netzteil und USB-Kabel ist eine üble Fehlerquelle! Das kenne ich nur zu gut vom ESP Zoo der hier bei mir so vor sich hin werkelt.

Doch in diesem Fall ist das sicher nicht die Ursache, da es mit anderen Version ohne Restarts läuft. Zudem gab es in der letzten Zeit mehrere Berichte mit ESP32 und Neustarts. Wenn du dir das Log anschaust wirst du sehen das der kleine Waldi(Watchdog) zugeschlagen hat und wenn auf einmal die Uptime nicht mehr zig Tage(bis zum nächsten FW update) ist sondern nur Stunden, dann liegt es mit extrem hoher Wahrscheinlichkeit an der FW. Damit die Entwickler hier nicht nur die Aussage bekommen "geht nicht" sonder auch gleich ein paar Daten zum schauen ist das Serial Log dabei.

Du hast aber Recht man kann nicht oft genug auf "miese" Netzteile und USB Kabel hinweisen!

lumapu, 2024-04-27:
danke für's Log, das ist wahr je mehr Hinweise um die einfacher lässt sich auf die Ursache schließen.

Du schreibst Limit Regelung, kann ich hierzu noch Details erfahren? 
* über MqTT oder API 
* interval (limit als auch Abfrage) 
* wie viele Inverter sind im Spiel 
* ist die Kommunikation sonst stabil 
* ist das Verhalten ohne Regelung identisch

blueline13, 2024-04-27:
Ja na klar hier die Infos.

- NUR das Limit selber wird über die API geschrieben.
- Alles was lesend ist, hole ich vom MQTT des ioBroker. Keine extra API Anfrage an die DTU.
- Wenn ich Limits über die API schreibe, dann ist der kürzeste Intervall 20 Sekunden. Warte auf das ACK von der DTU und dann nochmal so grob 12 Sekunden bis die Inverter das grob ausgeregelt haben.
- Sonst würde ich sagen das es nicht hakt, habe öfter ein Auge auf das WEB-Log und da kommen die Werte brav in dem Intervall an den man zum Lesen vorgibt. Auch die Weboberfläche verzögert nichts. Reagiert immer Flott.
- Ich habe bei einem Intervall von 10 Sekunden(unter Settings/Inverter der Wert) so grob 4-5 Neustarts pro Tag. Bei derzeit 5 Sekunden ist es grob das doppelte,  unten ein Überlick der Uptime der DTU vom iobroker, das du dir ein Bild davon machen kannst. Da gibt es auch keine fest Rhythmus in dem die DTU neu startet.
- Ich habe 4 Inverter 1xTSUN M1600(NRF), 2x HMS2000 und 1xHMS1000.
- Wenn die Abweichung vom Inverter größer ist dann werden mal alle 4 Inverter geschrieben über die API, bei kleinen Abweichungen ist es auch mal nur 1 Inverter.
- Mhhhh beim letzten Punkt .... die Limit Regelung ist immer aktiv solange Strom produziert. Ohne Regelung habe ich nicht getestet. Ist ein BKW mit 6,5kW😇 ich kann aber Morgen früh mal die Regelung abschalten, die DTU ist immer so zwischen 6-8 Uhr neu gestartet. Reiche ich dann morgen Vormittag nach.

Was ich sonst noch im Log so sehen konnte, das der Reset der DTU nicht wirklich direkt etwas mit den Limits zu hat. Die DTU startet auch neu wenn mehrere Minuten kein Limit geschrieben wurde. Wenn es hilft kann ich Morgen auch nochmal den ganzen Tag das log laufen lassen, mit abgeschalteter Regelung.
Das ist mein Ersatz ESP mit dem ich diese Test fahre und der normal produktive ESP läuft noch mit der 0.8.46, weil es da kein Limit Probleme gibt. Ich kann aber auch noch ein openDTU Fusion Board benutzen wenn es hilft.
Wollte aber mal wieder eine aktuelle Version testen. Vielleicht dazu noch das andere Ticket #1412 wo das merkwürdige Verhalten bei der Limit Regelung beschrieben ist. mit der aktuellen Version.

Ich hoffe das ich alle deine Fragen beantwortet habe, sonst frage gerne weiter.
[image]

lumapu, 2024-04-27:
danke für deine ausführliche Antwort, die hilft sehr weiter.
Wenn du ein Log liefern kannst wäre es noch besser, allerdings vermute ich, das auch hier nur die Waldis zu sehen sind. 
Nach dem ersten oder zweiten reboot kannst du getrost wieder mit der Limitierung weitermachen.
Ich hoffe den Fehler dann auch im Code identifizieren zu können, kann mir aber vorstellen dass der komplex sein wird.

blueline13, 2024-04-27:
Nur eine Sicherheitsfrage vorweg, welche Haken soll ich für das Serial Log anhaken?

lumapu, 2024-04-27:
serial Debug reicht aus. wichtiger sind die Meldungen vom ESP selbst

blueline13, 2024-04-28:
@lumapu 
Ich würde dir gerne ein Log geben aber die DTU rennt seit heute Morgen bis 13 Uhr in eins durch ohne einen Reset. Habe meine Regelung nun wieder aktiv. Scheint wohl wirklich an der Limit Regelung zu liegen. Mal sehen wie lange die jetzt läuft bis sie einen restart macht. Kaum nach dem die Regelung aktiv ist hat die DTU einen Neustart hingelegt. Soll ich die DTU noch ein paar Neustarts machen lassen und dir heute Abend das ganze Log von heute hochladen?

Wenn ich noch weitere Dinge testen soll, lass es mich wissen.

lumapu, 2024-04-28:
Jetzt müssen wir verstehen was die Regelung ändert, ich denke erst nach einer Änderung sind weitere Tests hilfreich. 
Die Details stehen glaube ich oben schon alle, falls ich noch eine Info brauche, melde ich mich

lumapu, 2024-04-29:
kannst du den Wertebereich nennen, den deine Regelung ausgibt - ich will ausschließen, dass evtl. negative Werte gesendet werden.

lumapu, 2024-04-29:
die negativen Werte können es nicht sein, das wird bei der Übernahme der Daten nicht ermöglicht und würde wenn nur den Wechselrichter stören

lumapu, 2024-05-08:
bitte nochmal mit der 0.8.117 testen. Auch die 0.8.116 ist möglich, diese läuft bei mir jetzt schon 3 Tage und 15h

blueline13, 2024-05-09:
Moin, die 116 lief seit dem Erscheinungsdatum und das Problem war noch vorhanden. Die 117 läuft und mal schauen was passiert.

blueline13, 2024-05-10:
Status 0.8.117: 
Leider kann ich nichts positives berichten. Habe noch immer mehrere Neustarts am Tag. Braucht ihr noch etwas, kann ich noch etwas testen?

lumapu, 2024-05-10:
Kannst du bitte prüfen, ob die 0.8.118 Besserung zeigt? Ich habe die Webserver Lib im Verdacht und habe gesehen, dass es eine neue Version gibt - diese wird jetzt verwendet.
Interasant wäre, ob die Reboots vor allem dann passieren, wenn die Regelung aktiv ist und du gleichzeitig auf den Webserver per Browser zugreifst. 

Gleichzeitig habe ich Ende April bei mir einen Fehler gesehen, der auch mit dem Webserver zu tun hat. Den Issue habe ich gerade angelegt und verlinke ihn hier zur Referenz:
https://github.com/esphome/ESPAsyncWebServer/issues/28

blueline13, 2024-05-11:
Moin, die 118 läuft. Ich habe fast immer ein zusätzliches Webbrowser Fenster auf die default Webseite offen. Das habe ich gerade geschlossen. Aber ich  habe nie wirklich etwas darüber gemacht, eher als Faulheit um nicht jedes mal den Link einzugeben.

Wenn ich mich recht entsinne dann hat das Arbeiten auf der Weboberfläche keinen wirklichen Einfluss auf die Reboots. Ich würde aus den Beobachtung eher sagen das mit der Häufigkeit zu tun hat wie oft Limits geschrieben werden. 

Zudem ist mir dabei etwas aufgefallen was sich eher komisch ausnimmt aber mit der 118 deutlich besser darstellt. Die letzten Tage war es mit einer extremen Häufigkeit das immer, meistens beim letzten Inverter, das Limit beim ersten schreiben überhaupt nicht angenommen wurde oder kein ACK kam. Meistens kam es dann beim zweiten-nten Versuch unverzüglich. Die neue Lib scheint wirklich an der Stelle etwas gebracht zu haben oder meine Inverter hatten ihre zicken Tage 😉  Wenn ich mir dein ESPHome Link anschaue, dann kann ich HEUTE ausschließen das ich einen zweiten Client habe der auf die DTU zugreift. Mal schauen was heute am Ende des Tages zu sehen ist. Eins was auch gestern aufgefallen ist, es hatte einen unterscheide gemacht in welchen zeitlichem Abstand die Limit geschrieben wurden. Ich hatte bis gestern Mittag eine Verzögerung von grob 100ms nach jedem Schreiben eingebaut. Aber das war irgendwie auch das Problem als ich es dann ohne Verzögerung geschrieben(nur die Laufzeiten im ioBroker zwischen 4-10ms) hatte lief es etwas runder und die Limits wurde etwas besser angenommen. Aber kein Vergleich zu heute, heute läuft es richtig guuut was die Schwuppigkeit beim Limit schreiben angeht! 😃

lumapu, 2024-05-11:
Das freut mich zu hören, wenn es weiterhin auftritt könntest du mich noch mit deinem Script versorgen, sodass ich es versuchen kann bei mir nachzustellen. 
Das zusätzliche Browserfenster macht schon was aus, da es ja nach Ansicht (Index oder live) im Hintergrund die Daten aktualisiert und damit die API aufruft.

blueline13, 2024-05-11:
Leider zeigt die 118 noch immer Reboots. Ich kann aber definitiv sagen das es mit der Häufigkeit beim Schreiben von Limit zusammenhängt. Einmal Bewölkung und das andere mal Lasten im Haus.

Na klar bekommst du das Script! Ich gehe mal davon aus das dir der Teil ausreicht wo die DTU angesprochen wird?

Der Teil liegt in einer Funktion die mit einem Limit in Watt gefüttert wird. Dann füge ich noch mal schnell ein paar Kommentare für das bessere Verständnis ein :wink: Das ganze ist in Blockly. Aber bitte um etwas Nachsicht, da es noch nicht wirklich fertig ist, da die DTU sich ja auch noch etwas verändert. Ballast raus und Übersicht rein kommen dann später 😇

dtuuser, 2024-05-11:
@blueline13
Wie oft und mit welchen Abstand schreibst Du die Limits? 
Ich schreibe alle 60 Sekunden, wenn nötig, das Limit. Komme so auf eine maximale Tageseinspeisung von knapp 10% vom Ertrag.

blueline13, 2024-05-11:
@dtuuser 
Ich glaube ich treibe die DTU & Inverter an ihre Grenzen wenn viel gereglt werden muss.  Es gibt einen 5 Sekunde intervall in dem die Lestungsabweichung geprüft wird, wenn keine Limits geschrieben werden. Ist die Abweichung grüßer als  -40 oder +10 Watt Bezug, wird als nächstes gerpüft ob das neue Limit pro Inverter mehr als 0.2% vom vorherigen % Wert abweicht um unnöiges Schreiben von Limits zu vermeiden. Wird ein Limit auf einen WR geschrieben wird geprüft ob das Limit auch akzeptiert wurde und danach werden Pausen eingefügt zwischen 8-15 Sekunden, die die Inverter für einen stabilen Betrieb brauchen um sich nicht abzuschalten wenn mal ein paar Limits hintereinander kommen. Im kürzesten Intervall können derzeit alle 10 Sekunden Limits kommen. Aber da der Webserver in der DTU derzeit etwas hakt, könnten vielleicht die Intervalle auch noch kürzer sein. Mal sehen was geht wenn alles schnurrt :joy: 

Ich galube das 6,5kWp BKW 😇 kann bis zu 55 kWh pro Tag liefern und ich komme bei mir auf eine Einspeisung dabei von grob 0,5-1kW pro Tag. Wobei nicht auf 0 sondern auf 20Watt Einspesung gereglt wird. Achja und dann wird auch noch demnächst die Temperatur der Inverter mit geprüft und werden dann auch dynamisch runter geregelt. Denn über 60°C im Inverter kann nicht gut sein für seine Elektronik auch wenn die gesamten Bauteile für den erweiterten Temperaturbeireich gewählt wurden. Die Inverter sollen ja schön lange halten 😉

blueline13, 2024-05-11:
@lumapu 
So ich habe das Script nun etwas besser kommentiert und ich hoffe ich habe nichts wichtiges übersehen. Sicher ist da mehr drin als nötig aber wie immer ist es "historisch gewachsen" 😂 und nimmt an vielen Stellen noch Rücksicht auf alte Versionen oder wo es mal rumpelte bei der Regelung. Auch sind noch Teile drin die nicht für das reinen Regeln zuständig sind sonder eher für das ganze drumherum das ich sonst noch benötige. Wenn du trotzdem noch was wissen möchtest, werde ich gerne versuche das besser zu beschreiben warum ich das mache. Sonst kann ich es auch versuchen weiter zu entschlacken. Lasse mich mal überraschen ob die Kommentare so reichen. Ich konnte bei mir keine Export Datei schreiben und habe es über die Zwischenablage kopiert.
[Limit Script.txt](https://github.com/lumapu/ahoy/files/15283322/Limit.Script.txt)

dtuuser, 2024-05-11:
@blueline13 
Oha. Vielen Dank für die vielen Infos. Ich prüfe alle 10 Sekunden den Stromzähler und regel dann runter oder rauf. Aber das nächste Limit setze ich erst nach frühestens 60 Sekunden. 
Mit der Temperatur habe ich noch garnicht auf der Liste. Aber irgendwie logisch. Heute hatte ich ein Max von 57,2 Grad. 
Bei einem Limit am WR von 20Watt, bekomme ich manchmal 40 Volt von den Solarzellen. Das kann ja auch nicht gut sein, oder?

blueline13, 2024-05-11:
@dtuuser 
Die 40 Volt sind für die Hoymiles Inverter keine Problem, da verkraften die bis zu 60 Volt und da ist noch genug Luft. Hohe Temperaturen erzeugt eine künstlicher Alterung der Bauteile und das macht Geräten viel mehr zu schaffen. Ich habe auch einen Inverter der abgesetzt an einer kühlen Wand in der Garage hängt. Der war unter Vollast auch über 70°C "warm" geworden 🙈 Also  ein Lüfter spendiert und dann war sein Maximalwert nur 45°C. Die 45°C sind für mich OK wo ich weiß der wird lange laufen. Ich glaube wir rutschen hier aber gerade ein wenig ab vom Hauptthema.

lumapu, 2024-05-25:
ab der 0.8.122 gibt es eine neue Möglichkeit den Fehlern auf die schliche zu kommen.
Der ESP 32 hat jetzt unter /system einen Button um Coredumps herunterzuladen.
Das schöne: man braucht keinen PC per USB verbinden und kann den Dump auch viel später noch laden. Der ESP speichert in seinem Dateisystem beim Absturz seinen Stacktrace.
Die geladene Datei ist erst mal unleserlich, mit der zugehörigen elf Datei kann man daraus jedoch rekonstruieren wo der ESP hängengeblieben ist. 

Falls ihr so einen Dump für mich habt könnt ihr mir diesen gerne zur Verfügung stellen (Discord oder per Mail contact@ahoydtu.de).

Bitte nur Coredumps senden, wenn der ESP mit der Version 0.8.122 oder neuer abgestürzt ist.

blueline13, 2024-05-26:
@lumapu 
Hallo, 
ich habe jetzt einige Reboots und einmal Stromlos hinter mir. Doch bei mir gibt es nur eine geschweifte Klammer als Download. 
[image]

lumapu, 2024-05-26:
das ist eigenartig, dann kann er bei dir nicht auf die Partition zugreifen. 
Welchen ESP und welche Installationsmethode hast du verwendet?

blueline13, 2024-05-26:
Ich habe einen normalen ESP32 und die letzten Versionen sind immer per OTA installiert worden. Ich glaube davor war das mit eurem Webinstaller und Erase gewesen.

Soll ich den mal auf eine bestimmte Weg neu flaschen und wenn ja kann ich dann ein Backup einspielen oder am besten alles neu einrichten?

lumapu, 2024-05-26:
im Prinzip hat du alles so gemacht wie ich auch. Daher wundert es mich, dass es bei dir nicht funktioniert. Hast du einen zweiten ESP? 
Ich könnte in der nächsten Version eine Fehlermeldung zwischen den geschweiften Klammern ausgeben.

blueline13, 2024-05-26:
Im Moment läuft die Test-DTU. Dann habe ich noch eine zweite wo noch die 0.8.46 drauf ist und ich kann auch noch einen openDTU Fuison Board dienen.

Am schnellsten geht es wenn ich mal eben meine zweite ESP32 DTU neu flashe..... melde mich nach dem nächsten Reboot 😄

blueline13, 2024-05-26:
@lumapu 
Habe dir gerade zwei coredumps per Mail geschickt. 

Bei der zweiten DTU habe ich euren neuen Webinstaller genommen und gleich die 0.8.122 genommen. Da stand der Dump sofort zur Verfügung.

lumapu, 2024-05-26:
echt eigenartig, dem müssen wir dann auch noch auf die Schliche kommen.
Danke ich werde es mir anschauen und hoffentlich was aussagekräftiges finden.

blueline13, 2024-05-26:
So habe nun die DTU die keinen coredump erzeugen wollte neu geflascht mit erase und siehe da, sie hat sofort einen ausgespuckt. Ich würde diesem Verhalten keine weitere Beachtung schenken da es sicher eine diesen berühmten Ausnahmen ist. Die ist ist ja bis zu 10 mal pro Tag neu gestartet. Da hat die sicher irgendwie, irgendwo, irgendwann einen mitbekommen 😉

lumapu, 2024-05-30:
bitte die aktuelle dev testen, hier hat sich was getan 😉 `0.8.123`

blueline13, 2024-05-31:
Doch Leider ist das Problem noch immer vorhanden 😞 Habe die cordeumps von heute hier liegen. Soll ich die dir wieder per Mail senden?

lumapu, 2024-05-31:
ja gerne - ich haben auch schon einen von Gubi, aber der hat nicht zur Fimrware gepasst. Hoffentlich ist es bei deinen besser 😉

lumapu, 2024-05-31:
die Coredumps von dir haben leider nur die Aussage, dass der Stack corrupt ist. Das heißt suchen, mal sehen wo sich dieser Fehler versteckt. Vielen Dank für die Hilfe mit dem Coredump - da weiß man immerhin bisschen was - aber leider halt nicht die exakte Stelle.
Hier ein Ausschnitt:

<details>

```
======================== THREADS INFO =========================
#0  0x40083cb4 in panic_abort (details=0x3ffbed06 "abort() was called at PC 0x401174f0 on core 0") at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
#1  0x4008c990 in esp_system_abort (details=0x3ffbed06 "abort() was called at PC 0x401174f0 on core 0") at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
#2  0x40092b40 in abort () at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
#3  0x401174f3 in task_wdt_isr (arg=<optimized out>) at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/task_wdt.c:176
#4  0x40085128 in _xt_lowint1 () at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:1114
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
Retrying reading threads information...

       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3ffbd4d4                 1073467588/0           76/384
0x3ffd0c4c                 1073468988/0        78420/384
0x3ffb38a4                 1073425556/18           76/480
0x3ffbb580                 1073455472/23           88/976
0x3ffb79f0                 1073444320/24           88/416
0x3ffb4888                 1073430136/20           80/528
0x3ffb5a00                 1073433072/19           88/432
0x3ffd7d40                 1073616380/3        43128/432
0x3ffb734c                 1073442620/24           84/416
0x3ffd79c8                 1073612268/1        39904/480
0x3ffb8188                 1073414944/1        24240/784
0x3ffddff8                 1073586152/3           80/448
0x3ffbcb08                 1073461496/22           80/416
0x3ffaf24c                 1073550332/23       146284/464

==================== THREAD 1 (TCB: 0x3ffbd4d4, name: '') =====================

==================== THREAD 2 (TCB: 0x3ffd0c4c, name: '') =====================

==================== THREAD 3 (TCB: 0x3ffb38a4, name: '') =====================

==================== THREAD 4 (TCB: 0x3ffbb580, name: '') =====================

==================== THREAD 5 (TCB: 0x3ffb79f0, name: '') =====================

==================== THREAD 6 (TCB: 0x3ffb4888, name: '') =====================

==================== THREAD 7 (TCB: 0x3ffb5a00, name: '') =====================

==================== THREAD 8 (TCB: 0x3ffd7d40, name: '') =====================

==================== THREAD 9 (TCB: 0x3ffb734c, name: '') =====================

==================== THREAD 10 (TCB: 0x3ffd79c8, name: '') =====================

==================== THREAD 11 (TCB: 0x3ffb8188, name: '') =====================

==================== THREAD 12 (TCB: 0x3ffddff8, name: '') =====================

==================== THREAD 13 (TCB: 0x3ffbcb08, name: '') =====================

==================== THREAD 14 (TCB: 0x3ffaf24c, name: '') =====================
```
</details>

lumapu, 2024-08-13:
wie verhält es sich bei den aktuellen Versionen?

blueline13, 2024-08-14:
Ich habe mittlerweile ein paar Veränderungen vornehmen müssen. Mein Zoo aus Invertern ist mal wieder um einen gewachsen so das es nun 5 Stück sind. Die Folge war das die Limitregelung überhaupt nicht mehr zu gebrauchen war, weil das Limit sehr oft 2-4 mal geschrieben werden musste bis es durch den ESP bestätigt wurde. Das hat manchmal bis zu zwei Minuten gebraucht. Also wurden die 5 Inverter auf zwei ESP (1 x 3 Inverter und 1 x 2 Inverter) aufgeteilt mit deutlich unterschiedlichen Frequenzen für das CMT Modul.

Damit war es dann wieder wie vorher auch und im üblich Bereich von 2-10 Neustarts pro Tag und die Regelung lief wieder deutlich besser. Das war bis einschließlich der Version 0.8.130. Eine Besserung gab es erst mit der Spezial Version von 0.8.135 die du als ZIP ((https://github.com/mathieucarbouESPAsyncWebServer @ ^3.1.3)  240811_ahoy_dev_esp32_0.8.135.zip ) hier veröffentlicht hast mit einem neuen Webserver. Die läuft erst seit zwei Tagen und bin noch vorsichtig mit Aussagen. Was ich derzeit sicher sagen kann der neue Webserver läuft besser. Was ich auch sagen würde ist das es deutlich mehr Erfolg beim setzen des ersten Limit gibt ohne es ein weiteres mal schreiben zu müssen. Zum zweiten ist der ESP mit zwei Invertern (2x HMS 2000) gestern das erste mal einen ganzen Tag durchgelaufen und das ist ein deutlicher Fortschritt!

Was mich aber immer wieder wundert ist wenn man Limits (50.0%) schreibt und es im ersten Anlauf nicht klappt, danach das Limit um 0.1% erhöht(50.1%)und neu schreibt, wird oft das Limit vom ersten Versuch bestätigt also die 50.0%. Ist das irgendwie erklärbar?

Auch habe ich mal geschaut wenn das Limit nicht durch den ESP bestätigt wird, mal zu warten wann die Bestätigung kommt. Ich hatte mal 30 Sekunden auf die Bestätigung gewartet und die war dann noch immer nicht da. Darauf hin habe ich Zyklus für das schreiben eines nicht bestätigten Wertes deutlich verkürzt auf 7 Sekunden. Denn wenn das Limit nicht nach 1-3 Sekunden bestätigt wurde dann kam auch nach fast 30 Sekunden keine. Das ist deutlich aggressiver aber die Inverter überleben das scheinbar und ich hatte auch auch noch keinen Neustart der Inverter wegen des Vorgehens gesehen. Das war mit "alten" Versionen immer mal wieder der Fall gewesen.

Was auch derzeit zu beobachten ist der ESP mit 2 Invertern läuft irgendwie etwas runder als der mit 3 Invertern. Also hat die Anzahl der Inverter einen gewissen Einfluss in welcher Form auch immer.

Ich hoffe das hilft dir erst mal als Info.

lumapu, 2024-08-16:
> Eine Besserung gab es erst mit der Spezial Version von 0.8.135 die du als ZIP  ...

Dieser Webserver kommt jetzt in der development Version beim ESP32 zum Einsatz, bin gespannt welches Feedback die große Runde der Tester gibt.

> Was mich aber immer wieder wundert ist wenn man Limits (50.0%) schreibt und es im ersten Anlauf nicht klappt, danach das Limit um 0.1% erhöht(50.1%)und neu schreibt, wird oft das Limit vom ersten Versuch bestätigt also die 50.0%. Ist das irgendwie erklärbar?

für mich nicht erklärbar

blueline13, 2024-08-22:
@lumapu 
Jetzt habe ich mir das wirklich sehr lange angeschaut wie sich das ganze verhält.

Punkt1:
Status zu 0.8.129 & 0.8.141

Das einzige was ich zu den ganzen Version allesamt wirklich sicher sagen kann, dass die Anzahl der geregelt Inverter pro DTU für die Neustarts verantwortlich ist!

Bis 2 Inverter pro DTU geht es ohne einen einzigen Neustart und das über Tage hinweg!!!

Ab dem 3 Dritten geregelt Inverter sind es dann so 2-n Neustarts pro Tag.

Zudem leidet ab dem 3ten Inverter auch das erfolgreiche Schreiben im ersten Anlauf (Subjektiver Eindruck). Ich habe eben mal einen Counter eingebaut wie oft das so pro Tag vorkommt.

Punkt 2:
Grundsätzliche Stabilität beim schreiben des Limits, wie es hier in den letzten Tagen mit der 0.8.141 berichtet wurde. Das ganze kann ich so nicht nachvollziehen. Nachdem ich meine Fehler beseitigt hatte. Aus welcher auch immer nicht nachvollziehbarem Grund, hatte ich geschafft meine F-Box auf Autoselect beim WLAN Kanal zu stellen und die F-Box hatte sich natürlich Kanal 13 genommen. Also fix auf einen festen und siehe da, damit waren die ersten kompletten Timeouts verschwunden. Zudem hatte ich parallel im ioBroker von exec und curl auf HTTP Post umgestellt. Danach verschwanden die Timeouts fast gegen 0 (5-10 Timeouts in 2 Stunden).

Ich würde von meinem Standpunkt aus keinen wirklich Unterschied zwischen dem alten und neuen Webserver sehen. Derzeit läuft auf einer DTU die 0.8.129 und auf den anderen 0.8.141. Werde heute Abend mal die beiden Versionen tauschen um zu sehen ob das Resend des Limit auf der FW Version zurück zu führen ist.

Wenn ich etwas testen soll, dann lass es mich wissen.

blueline13, 2024-08-23:
@lumapu 
Hier ist die Tabelle und ich glaube die Tendenzen sind sehr eindeutig. Ich hoffe es hilft euch. Wie du dir sicher denken kannst, werde ich bei mir auf die 0.8.129 zurück gehen. Scheinbar bringt der neue Webserver nicht das was du gehofft hattest. Wenn ich noch andere Versionen testen soll um sie zu vergleichen, sag einfach Bescheid.

Das deckt sich so halbwegs ja auch mit einigen anderen Berichten. Was ich so sehen konnte ist ebenfalls so, das die Anzahl der Inverter sehr deutlich einfluß auf die Stabilität bei der Limitsetuerung hat. 

Ich würde eher Wert darauf legen das das Schreiben des Limt sauber funktioniert auch wenn die DTU dann ein paar mal pro Tag neu startet. 

Wie wäre es mit der Idee, ein Array zu benutzen um alle Werte mit einmal zu schreiben?

[image]

blueline13, 2024-08-25:
Noch ein Nachtrag zu den FW und den Resend Limit. Das Problem ging definitiv mit der 0.8.130. Da gibt es nur eine Änderung #1704 die sich auf Power Limit bezieht.

[image]

lumapu, 2024-08-29:
@blueline13 schau auch in 

* #1726

dort geht es auch um Regelung. Ich hoffe die `0.8.143` ist wieder auf dem Niveau von der `0.8.129`, nur schneller.

blueline13, 2024-08-30:
Moin,

Ich hatte eben die 0.8.143 auf meinen DTUs. Ich kann leider nicht wirklich etwas positives berichten.

Das rebooten bleibt und das Resend von Limts erreichte nach 3 Stunden schon deutlich mehr als 0.8.129 nach einem ganze Tag und bin eben wieder zurück auf 0.8.129. Seit dem habe ich in 20 Minuten nur 6 Resends gehabt bei 5 Invertern vorher mehrere pro Minute über alle Inverter und sogar mehrfach hintereinander bis dahin das zwei Inverter komplett auf 0 Leistung gingen und einen Reboot benötigten. Keine Ahnung ob das nun Zufall war eben aber was sich abzeichnete, dass der letzte Inverter in der DTU Liste, wieder deutlich mehr Resends hatte als die davor.

Hast du eine Info wie das hier funktioniert(improved sending limits of multiple inverters in very short timeframe #1726)? Gibt es etwas an Infos was das Timeframe sein soll? Hatte mal ohne Verzögerung zwischen den Limits laufen, mal mit 50 ms, 100ms, 150 ms. So einen wirklich unterschied konnte ich nicht feststellen bei der 0.8.143. "Gefühlt" ist 0.8.143 die schlechteste seit der 0.8.129. Bei der 0.8.129 kann ich ohne zeitlichen Verzögerungen arbeiten. Und was da verloren geht würde ich eher dem WLAN allgemein zuschreiben. Ich muss mir mal das Ethernet Modul für das OpenDTU Fusion Board holen um zu sehen ob es daran liegt und eine angebunden DTU kann nicht so schnell weglaufen 🤣 

Da ist eben der Vorschlag alle Limit für eine DTU mit einmal zu schreiben, vielleicht bringt das etwas Linderung auch wenn es nicht Ursache beseitigt. Sicher wird es da draußen nur 1% der Anweder betreffen die mehr als zwei Inverter regeln. Was man in anderen Tickets so lesen kann. Für Zwei Inverter ist die 0.129 eine echt gute Version. Ab dem dritten Inverter kommen dann die Reboots dazu.

blueline13, 2024-10-04:
@lumapu 
Nach dem die neue Versio 0.8.151 wieder gut mit den Limits läuft würde ich dir gerne erneut ein paar Coredumps von den heutigen reboots geben. Vielleicht findet sich da doch irgendwo ein Hinweis warum das passiert.

[2024-10-04_11-35-22_v0.8.151_esp32-wroom32_coredump.zip](https://github.com/user-attachments/files/17256323/2024-10-04_11-35-22_v0.8.151_esp32-wroom32_coredump.zip)

blueline13, 2024-10-04:
@lumapu 
Noch eine Erweiterung. Da jetzt scheinbar ein paar mehr von den Reboots betroffen sind habe ich mal mein Fusion Board mit Ethernet ran gehängt und da passiert fast ähnliches. Im Bild ist das letzte Absacken des Heaps vom Fusion Board(die ersten 4 sind von dem ESP32) zu sehen und das verhält sich auch anders. Vermutlich weil es kein WLAN mehr nutzt und dafür auch mehr Speicher hat. 

Diesmal passiert folgendes. von Der DTU kommen keine Daten mehr per MQTT zurück, dass Webinterface ist tot, keine API und nur ein Ping auf die IP geht noch. Gerade fehlt mir etwas die Zeit sonst würde ich nochmal per Serial und Notbook mitloggen was dann zu sehen. Da die DTU nicht neu bootet gibt es auch kein Coredump. Die ESP mit Wlan booten vermutlich deswegen weil der Watchdog vom WLAN nicht mehr getriggert wird.

[image]

blueline13, 2024-10-04:
@lumapu
Noch ein Nachtrag, ich hatte eben ausprobiert was passiert wenn der Speicher immer weniger wird und ich die Limit Regelung stoppe. Dann stoppt sofort der Speicherverbrauch und bleibt da konstant hängen. Startet die Regelung wieder, wird der Speicher sofort wieder weniger. Diese Reboots müssen also schon etwas mit der Limit Regelegung der DTU zu tun haben.

lumapu, 2024-10-04:
das ist sehr hilfreich, ich bin schon am überlegen was das sein könnte

blueline13, 2024-10-04:
Also haben die coredumps wieder nicht wirklich helfen können. Schade. Ich kann gerne nochmal das Fusion Board (ich wäre erst mal vorsichtig mit dem LAN und WLAN vom Fusion Board, meine auch das sich damals auch mal ohne Ethernet so verhalten und nichts mehr per MQTT zurück kam. Daher werde ich das nochmal prüfen) ohne Ethernet laufen lassen und dann per USB mitloggen was ich dort im Log dann finde. Als sich heute das Fusion Board verabschiedet hatte würde ich sagen das es noch fleißig weiter gefunkt/gestört hatte, obwohl es keine Daten von mir bekommen hatte. Denn als ich den ESP32 wieder parallel gestartet hatte, gab es soviel Fehler in der Übertagung und erst als das Fusion Board stromlos war lief der ESP rund. Aber ohne Webserial ... Ich werde aber vermutlich erst am Sonntag dazu kommen.

Falls ich sonst noch irgendwie helfen kann, was testen oder oder oder .... mache ich jedenfalls gerne um allen zu helfen, denn das Rebooten scheint nun deutlich mehr Leute zu betreffen.

blueline13, 2024-10-06:
Moin. Jetzt habe ich vom Fusion Board eine Fehlermeldung bekommen. Was mich jetzt mächtig ins grübeln bring, wie kann etwas mit WiFi kommen wenn da ein Ethernet Board drauf steckt. Da ist die einzige Meldung gewesen die ich bekommen hatte. Ich kann mich erinnern das das in etwa wie im Weblog ausgesehen hatten mit den Meldung. Egal hier erst mal die Meldung:

13:09:22.637: [12730898][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"
[12730907][E][WiFiClient.cpp:429] write(): fail on fd 48, errno: 11, "No more processes"

blueline13, 2024-10-07:
Moin,

also mit dem Ethernet Board gibt es keinen Reboot wenn er Amok läuft!

Mhhhh wenn ich das Ethernet Board runter nehme, startet er nicht mehr und es gibt folgende Meldung. Siehe Ende. 

Wenn beim Amoklauf das Board nicht per Watchdog neu bootet muss er noch was machen und in einem Loop hängen und laufen. Deckt sich mit einem zweiten ESP der dann extrem viele fehlerhafte Pakete hat, bis man dem Amokläufer den Saft abdreht. Also wird der Watchdog noch getriggert, wenn ihr den Hardware Watchdog des ESP benutzt.

Ansonsten, wenn ich meinem Gefühl folgend würde und mir alle offene Tickets so anschaue, insbesondere alles was mit MQTT zu tun hat würde mir meine Intuition sagen, irgendwo hapert es mit dem MQTT wenn es Limits gibt. Ich meine es gibt ein Ticket wo gesagt wurde wenn MQTT abgeschaltet wurde dann läuft seine Regelung wieder. Was sich auch zeigt wenn es nur wenige Limits gibt, dann scheint es deutlich länger zu laufen. Wenn du mal etwas testen willst um zu sehen ob der Heap weniger wird, würde ich der DTU alle 15 Sekunden ein neues Limit verpassen, so im Rahmen von 100% -> 80% in 0,3% Schritten. Sollte bei dem derzeitigen Wetter ja passen. Bei meinen 5 Invertern geht das natürlich schneller. Als ich mal die Inverter auf zwei ESPs aufgeteilt hatte, lief der mit 2 Stück durch und der mit 3 hatte gebootet. Bin mir aber nicht mehr sicher, da schon länger her. Wenn gewünscht kann ich das nochmal umbauen und testen.

Wenn nur wenige Limits kommen, scheint es ja immer eine Zeitlang zu laufen und der Heap bleibt auf dem gleichen Niveau.

Ich würde ja gerne mal das ganze ohne MQTT laufen lassen aber das wird noch etwas dauern, ich versuche gerade mit der API aus meinem ioBroker mit Blockly alles zu ersetzen was MQTT liefert und ich so brauche/aufzeichne.

Fehler:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420cba92
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
06:06:43: E (1346) w5500.mac: w5500_reset(230): reset timeout
E (1346) w5500.mac: emac_w5500_init(715): reset w5500 failed
E (1347) esp_eth: esp_eth_driver_install(214): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x4037d0f4
file: "network/AhoyEthernetSpi.h" line 99
func: void AhoyEthernetSpi::begin(int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)
expression: esp_eth_driver_install(&eth_config, &eth_handle)

abort() was called at PC 0x4037d0f7 on core 1

Backtrace: 0x40377da2:0x3fcebc80 0x4037d101:0x3fcebca0 0x40383c0d:0x3fcebcc0 0x4037d0f7:0x3fcebd40 0x4201ef4d:0x3fcebd60 0x4201f0a9:0x3fcebe30 0x4201f82e:0x3fcebe60 0x42022a0d:0x3fcebf20 0x4203a506:0x3fcebf40

ELF file SHA256: 3df24a6583be5143

Rebooting...

lumapu, 2024-10-07:
Der Fehler kommt, wenn Ethernet konfiguriert ist, aber nicht verfügbar ist. Das wird bisher noch nicht angefangen.
Im es zu umgehen einfach das Ethernet Board wieder aufstecken und dann in den Einstellungen den Haken bei `enable` entfernen. Danach die es auch ohne gehen, bzw. es ist egal ob gesteckt oder nicht.

stefan123t, 2024-11-02:
@lumapu bitte schau Dir mal die issues zum ESPAsyncWebServer von mathieucarbou auf der OpenDTU bzw. OpenDTU-OnBattery Seite durch. Z.B. hier https://github.com/tbnobody/OpenDTU/issues/2314#issuecomment-2383227247

Prinzipiell ist der Code von mathieucarbou m.E. um einiges aktueller und besser als der aus dem espressif / me-no-dev repo. Andererseits hat Mathieu erst kürzlich eine API Änderung mit der Einführung der AuthenticationMiddleware die noch einige Ecken und Kanten hatte. Unter anderem wurde häufig eine neue AuthenticationMiddleware instanziiert, obwohl die bestehende evtl. genauso gut gewesen wäre.