lumapu/ahoy #1602: Inverter Interval

slg60, 2024-04-30:
Gibt es eigentlich irgendwelche Risiken/Nachteile wenn man das Inverter Interval auf z.B 3 Sekunden setzt? 5 Sekunden liefen bei mir bisher problemlos bei einer RX Fail Rate von unter 5%, wollte jetzt auf 3 Sekunden runter weil die Stromwerte sehr zügig kommen und es dann bei schnellen Änderungen zu Berechnungsfehlern kommt. Ich nutze 2 HMS16004t mit dem CMT2300 und einem ESP32.

knickohr, 2024-04-30:
Beobachte mal das Webserial ob die Loop-Zeiten innerhalb der Intervallzeiten sind. Wenn nicht, isses zu eng und kann zu Risiken, Hautausschlägen, Pickel und Nebenwirkungen führen 😉

slg60, 2024-04-30:
Irgendwie hatte ich beim schreiben schon die Vermutung das ich für die Frage hochgenommen werde🤣

slg60, 2024-04-30:
In der Webserial passt das alles zusammen und die Daten kommen auch alle 3 Sekunden, sehe ich auch an den mqtt Daten die geloggt werden. Ich dachte jetzt eher so an das Funkmodul und den esp. CPU Last und CPU temp kann man ja leider nicht sehen, wobei ich dem esp das eigentlich zutraue...

knickohr, 2024-05-01:
Das mit den Loop-Zeiten war ernst gemeint !

rmayergfx, 2024-05-01:
Bitte dabei auch beachten noch einen Sicherheitspuffer mit einzubauen ! Es kann ja immer mal wieder zu Packet Loss kommen und dann schwanken die Werte. 18529ms, wow, wieso ist denn der Wert so hoch bei dir?
Habe hier
I: com loop duration: 176ms
I: com loop duration: 766ms
zwischen diesen beiden Werten schwankt es, AhoyDTU ist auch nur 5m vom WR entfernt.

slg60, 2024-05-01:
Loop Zeit schwankt bei mir zwischen 600 und 850ms

0.8.101 - 2a4c836
---------------
10:22:53.324 I: (#0) RX  83ms | 27 -38dBm | 95 01
10:22:53.325 I: (#0) RX 132ms | 27 -38dBm | 95 02
10:22:53.326 I: (#0) RX 182ms | 27 -38dBm | 95 03
10:22:53.327 I: (#0) RX 232ms | 27 -38dBm | 95 04
10:22:53.329 I: (#0) RX 273ms | 15 -38dBm | 95 85
10:22:53.331 -----
10:22:53.656 I: (#1) RX  84ms | 27 -58dBm | 95 01
10:22:53.657 I: (#1) RX 133ms | 27 -58dBm | 95 02
10:22:53.658 I: (#1) RX 183ms | 27 -58dBm | 95 03
10:22:53.659 I: (#1) RX 233ms | 27 -58dBm | 95 04
10:22:53.660 I: (#1) RX 274ms | 15 -58dBm | 95 85
10:22:53.672 -----
10:22:53.673 I: com loop duration: 673ms
10:22:53.674 -----
10:22:56.324 I: (#0) RX  83ms | 27 -38dBm | 95 01
10:22:56.326 I: (#0) RX 132ms | 27 -38dBm | 95 02
10:22:56.326 I: (#0) RX 182ms | 27 -38dBm | 95 03
10:22:56.327 I: (#0) RX 232ms | 27 -38dBm | 95 04
10:22:56.328 I: (#0) RX 273ms | 15 -38dBm | 95 85
10:22:56.341 -----
10:22:56.667 I: (#1) RX  84ms | 27 -58dBm | 95 01
10:22:56.668 I: (#1) RX 134ms | 27 -58dBm | 95 02
10:22:56.669 I: (#1) RX 184ms | 27 -58dBm | 95 03
10:22:56.670 I: (#1) RX 234ms | 27 -58dBm | 95 04
10:22:56.671 I: (#1) RX 274ms | 15 -58dBm | 95 85
10:22:56.683 -----
10:22:56.684 I: com loop duration: 684ms
10:22:56.685 -----
10:22:56.324 I: (#0) RX  83ms | 27 -38dBm | 95 01
10:22:56.326 I: (#0) RX 132ms | 27 -38dBm | 95 02
10:22:56.326 I: (#0) RX 182ms | 27 -38dBm | 95 03
10:22:56.327 I: (#0) RX 232ms | 27 -38dBm | 95 04
10:22:56.328 I: (#0) RX 273ms | 15 -38dBm | 95 85
10:22:56.341 -----
10:22:56.667 I: (#1) RX  84ms | 27 -58dBm | 95 01
10:22:56.668 I: (#1) RX 134ms | 27 -58dBm | 95 02
10:22:56.669 I: (#1) RX 184ms | 27 -58dBm | 95 03
10:22:56.670 I: (#1) RX 234ms | 27 -58dBm | 95 04
10:22:56.671 I: (#1) RX 274ms | 15 -58dBm | 95 85
10:22:56.683 -----
10:22:56.684 I: com loop duration: 684ms
10:22:56.685 -----

knickohr, 2024-05-01:
Den doppelten bis dreifachen Wert würde ich anpeilen. Vor allem zum Start der Inverter bei Tagesbeginn kann schon wegen der vielen Payloads (Gridprofil, Alarmdaten, bis die Heuristik einrastet, viele Timeouts und Retransmits produzieren.

@rmayergfx  Vielleicht mal die Anzahl der Inverter beachtet ?

slg60, 2024-05-01:
> Den doppelten bis dreifachen Wert würde ich anpeilen. Vor allem zum Start der Inverter bei Tagesbeginn kann schon wegen der vielen Payloads (Gridprofil, Alarmdaten, bis die Heuristik einrastet, viele Timeouts und Retransmits produzieren.
> 
Na dann liege ich ja mit meinen 3 Sekunden ja ziemlich Safe im grünen Bereich. 👍🏻

slg60, 2024-05-15:
Was würde eigentlich passieren wenn das Intervall zu klein gewählt wird?

knickohr, 2024-05-15:
Die Queue läuft über und irgendwann rebootet die DTU.

slg60, 2024-05-15:
Ok, ich habe mal etwas länger protokolliert und hin und wieder habe ich da Ausreißer gesehen wenn mal ein tx Fall passiert dann liegt die Loop duration mal bei 5000ms. Das passiert aber nur ein mal und fängt auch dann wieder. Aber evtl ist das der Grund für meine reboot Problematik aus dem anderen Bug Report.

knickohr, 2024-05-15:
Die Queue hält einiges aus, darf auch mal drüber gehen. Aber wenn es permanent passiert macht es irgendwann die Krätsche.

slg60, 2024-05-15:
Ok, danke, echt schwierig dem Problem auf die schliche zu kommen😞

DanielR92, 2024-06-29:
Da es hier keine weitere Diskussionen ergibt, close ich das hier mal. 
Bitte neu öffnen wenn ich falsch gehandelt habe. :)

closed 2024-06-29 by DanielR92