lumapu/ahoy #1603: [Bug] No Internet but WLAN - DTU not working

jschwed, 2024-05-01:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

idontknow

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Hoymiles HM 800 
NRF24L01
invertert: intervall set to 15 sec

### Error description

currently we don't have an internet connection
local WLAN is fine, router+access point are working. I can access local devices within home WLAN

when checking http://ahoy-dtu/live the data shown there is outdated/ not updated

I can access DTU/ahoy settings incl. ie login, settings, system...
seems like the the inverter/ DTU needs an internet connection to visualize inverter live data, which is strange. It should be local, right?

Any further questions/ info I need to provide, please let me know.

jschwed, 2024-05-02:
thanks! 
you're right, system time is 1970-01-01 ....
I tried your solution, but it doesn't work.

a) when changing the settings, system time is shown correct, but polling does not start.
b) when changing the settings, save and reboot --> system time is again at 1970-01-01

Gubi2023, 2024-05-02:
ist das nicht dieses Fehlerbild?  https://github.com/lumapu/ahoy/issues/1593#issuecomment-2069639290

jschwed, 2024-05-02:
@Gubi2023 : I don't see it's the same error. WLAN (local network) is fine and DTU logged in as-well, ie I could login into settings.

@MetaChuh : thanks. Internet is back since this morning. I would prefer not to go to a beta version this time, but will check with the next release.

Thanks for your quick responses! Great experience. 
From my pov this bug can be closed, but can also keep it open if you need a final test from my side with the next release.
Best regards!

jschwed, 2024-05-03:
@MetaChuh : sure, will do. Looking forward to the next release! :-)

closed 2024-08-16 by lumapu