lumapu/ahoy #1604: Shutdown by remote controll

jofido, 2024-05-03:
## Hardware
  * [ ] ESP8266
  * [x ] ESP32
  * [x ] CMT2300A

hallo zusammen,

ich habe seit 2 Wochen meine Ahoy DTU in Betrieb und aktuell das Verhalten, das sich
der Inverter sporadisch abschalten. Als Fehlermeldung wir
124 Shutdown by remote control ausgegeben.
der Inverter kann dan über "TURN ON" wieder gestartet werden.
in der FritzBox habe ich vor dem "Shutdown" Eintrag immer eine Ab/Anmeldevorgang der AHOY DTU

Hat jemand eine Idee ?

tictrick, 2024-05-03:
Ja, Du hast vermutlich über MQTT einen retained topic der dir den shutdown bringt.

jofido, 2024-05-03:
Danke für den Tip
Wenn ich beim MQTT Broker die Einstellung "Eigene States beim Verbinden publizieren" abwähle tritt beim
Neustart bzw. beim Netzwerkaufba kein Shutdown mehr auf. 
Ich werde das mal noch beobachten.

closed 2024-08-16 by lumapu