lumapu/ahoy #1605: [Bug]V0.8.116: added max inverter temperature leider nicht komplett sichtbar

Gubi2023, 2024-05-05:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.116

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update letzte dev

### Error description

 screenshot PC
[image]

Screenshot I-Phone
[image]

Loetnase, 2024-05-05:
Könnte man die maxtemperatur nicht einfach unter die aktuelle Temperatur schreiben, ohne Datum und ohne Sekunden? Ich gehe davon aus das ist die maximale Temperatur des Tages, nicht sei letztem dtu Start. In etwa so wie auf meinem Bild unten dran.
[image]

Gubi2023, 2024-05-09:
nicht perfekt, aber besser...
[image]

closed 2024-05-09 by Gubi2023