lumapu/ahoy #1606: Reboot Schleife wenn Option "Werte bei Sonnenaufgang zurücksetzen" aktiv

mathiaslehmann, 2024-05-06:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.116

### Github Hash

e4a467c :: 0.8.116

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- 1 Wechselrichter
- Werte und Gesamtertrag um Mitternacht zurücksetzen aktiviert
- Werte bei Sonnenaufgang zurücksetzen aktiviert
- Maximalwerte mitternachts zurücksetzen aktiviert
- Grid-Profil auslesen
- MQTT aktiv
- Display SH1106 128x64 (1.3")

### Error description

Es kommt zu einer Reboot Schleife wenn die Option "Werte bei Sonnenaufgang zurücksetzen" aktiv ist.

lumapu, 2024-05-06:
kommt die Rebootschleife sofort oder erst bei Sonnenaufgang? Ich habe die Option bei mir auch aktiv, meine DTU hat eine Uptime von 1 Tag und 11 Stunden.

mathiaslehmann, 2024-05-06:
Das passiert ab Aktivierung der Option. Habe das immer im laufe des Tages probiert. Nach dem Speichern der Einstellungen und Reboot wird noch die Zeit aus dem Netz gezogen und dann erfolgt der erneute Reboot. Würde sagen sobald die erste Kommunikation zum Wechselrichter erfolgt oder erfolgen soll.

mathiaslehmann, 2024-05-06:
Habe es gerade mal aktiviert, dann läuft es erstmal. Mal schauen was morgen passiert. Ich werde berichten.

lumapu, 2024-05-06:
danke für den Hinweis. Ich habe es auch in der Nacht aktiviert, während keine Kommunikation aktiv war. Weiterhin wundert mich der Zusammenhang, aber evtl. wissen wir morgen mehr.

mathiaslehmann, 2024-05-07:
Guten Morgen, der Reboot hat heute morgen wieder begonnen, vermutlich mit Start der Kommunikation zum WR. Lassen Sie es mich wissen wenn ich noch irgendwas zur Fehlerfindung beitragen kann. Habe die Einstellung wieder deaktiviert, dann läuft die DTU durch.

lumapu, 2024-05-08:
hängt am gleichen Problem wie #1607 (war einfacher zu tiggern)

mathiaslehmann, 2024-05-11:
Vielen Dank.

closed 2024-05-11 by mathiaslehmann