lumapu/ahoy #1607: ESP32 116 bootet um 0 Uhr

dtuuser, 2024-05-07:
Moin zusammen,

mein ESP32 mit der DEV 116 bootet um Mitternacht, obwohl der Haken nicht gesetzt ist. 
Auch ein Setzen --> Neustarten --> Wieder wegnehmen --> Neustarten  hat nicht geholfen.

mellow5, 2024-05-07:
Ich schließe mich dem an. Ich habe es schon seit der 0.8.115 bemerkt.

jimknopf63, 2024-05-07:
Kann ich nicht bestätigen, meiner läuft seit 18 Std. mit DEV 116. Das entspricht der Zeit des Updates. Auf zwei ESP32 in verschiedenen Haushalten.

dtuuser, 2024-05-07:
Bei mir auch nur NRF24 mit externer Antenne und Display. Bei einer anderer DTU ohne Display mit 113 keine Probleme.

kiu77, 2024-05-07:
Bei mir wird um Mitternacht auch neugestartet (oder die Uptime falsch angezeigt). Es sollten nur die Maximalwerte um Mitternacht gelöscht werden.

[image]

Loetnase, 2024-05-07:
[image]
Kann ich nicht bestätigen, meiner lief seit gesten Morgen durch, zumindest laut Uptime.
ESP32 mit ePaper und NRF24 :thinking:

dtuuser, 2024-05-08:
Bei mir auch nach Re-Update. Reboot um 0 Uhr

GrisuXX, 2024-05-08:
Dito, auch bei mir um 0Uhr ein reboot.
Mit ESP32, NRF24 und obigen Einstellungen

Loetnase, 2024-05-08:
Da muss ich euch enttäuschen, bei mir kein Reboot, bei drei parallelen Systemen 
[image]
schaut auch mal hier 
https://discord.com/channels/984173303147155506/1075313280312152156/1237678682261291028
da stehen alle meine Einstellungen

Loetnase, 2024-05-08:
Passt das hier gerade vom Discord da auch irgendwie mit ins Bild
[image]

kiu77, 2024-05-08:
@Loetnase 
Mit Einstellungen meinte ich die bei den Invertern...

Loetnase, 2024-05-08:
Okay, das passt.
Ich will und brauche ja auch keinen Reset. 
Wenn ich die alten Werte sehe bis der Inverter aufwacht ist das für mich gut.

Dirk-Ahoy, 2024-05-08:
Hab das gleiche Problem - alle DTUs mit .116 rebooten püntktlich um 0:00  Habe nur Werte und Gesamtertrag um Mitternacht zurücksetzen aktiv.

sumerland, 2024-05-08:
Gleiches Problem mit selbstgebauter DTU, NRF24 und ohne Display. Habe nun auch alle Haken bzgl Wertereset rausgenommen. Werden die eh genullt wenn der Inverter aufwacht?

Loetnase, 2024-05-08:
@sumerland 
Das musst du anders rum sehen. Wenn der Inverter einschläft bleiben die letzten Werte in der DTU stehen z.B. Tagesertrag bis am Morgen der Inverter wieder aufwacht und neue Werte schickt welche mit 0 beginnen. 
Wenn du nur die Werte anschaust (wie ich) ohne irgendwelche Auswertungen dahinter ist das passend. Wenn jemand z.B. Tageswerte oder Maxwerte  abspeichert macht es Sinn die um Mitternacht zu nullen. Oder bei Batterieeinspeisung wenn der Inverter dauernd Strom hat und nie schlafen geht.

dtuuser, 2024-05-09:
Topp. Kein Reboot mehr.
Vielen Dank

GrisuXX, 2024-05-10:
Heute Nacht kein Reboot mit der 0.8.117 👍

closed 2024-05-10 by dtuuser