lumapu/ahoy #1608: [Bug] 8266 always starts in AP-mode since 0.8.112

IgelCampus, 2024-05-07:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.111

### Github Hash

388cd89

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard single inverter setup

### Error description

When updating to any version after 0.8.111 the AHOYDTU AP-mode is used instead of the configured WIFI connection.
Just tried this with 0.8.115 and 0.8.116 again with the same result. The last reliable version was 0.8.111 where the WIFI connection is established pretty much immediately after booting.

FYI: I am using several Mikrotik cAP ac access points for the WIFI.

IgelCampus, 2024-05-07:
I modified the Mikrotik configuration to always use channel 11 but unfortunately ahoy is still always in AP mode after a reboot. When completing the wizard, ahoy connects to the correct wifi network and is reachable as expected. But after the next reboot the game starts again.

lumapu, 2024-05-11:
will work on this topic next, thanks for confirming.

cibis8, 2024-05-20:
Bei mir das gleiche.. ab der 111 kann ich keine statische IP mehr definieren... er nimmt sich eine aus dem DHCP Bereich.

lumapu, 2024-05-22:
sorry forget about this one, need to focus on this ... please remind me if I do not respond within one week

lumapu, 2024-05-30:
is it related to static IP configuration? As I tried now with 0.8.122 I can't see the issue here. Will now try with static IP.
For static IP there is another issue around here.

cibis8, 2024-05-30:
Nach dem Update auf eine Version höher 111 kann ich nicht mehr eine feste IP eingeben. Geht nur noch via DHCP.

IgelCampus, 2024-05-31:
I can also report success after upgrading from 111 to 123. Restart and Wifi reconnect just worked fine.

IgelCampus, 2024-06-01:
Unfortunately the fix is unstable and reboots the 8266 every few minutes.

closed 2024-08-16 by lumapu