lumapu/ahoy #1610: [Bug] curl ahoy-dtu

MaDill, 2024-05-10:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

GIT SHA: 5ebfe5a :: 0.8.83

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

{"version":11,"wifi":{"ssid":"xxx","pwd":"","ap_pwd":"esp_8266","hidd":false,"dev":"AHOY-DTU","adm":"","prot_mask":61,"dark":false,"reb":false,"region":0,"timezone":1,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"cs":15,"ce":0,"irq":2,"sclk":0,"mosi":0,"miso":0,"en":true},"ntp":{"addr":"pool.ntp.org","port":123,"intvl":720},"sun":{"lat":xxx,"lon":xxx,"offs":0,"offsEve":0},"serial":{"show":false,"debug":false,"prv":true,"trc":false,"mqtt":false},"mqtt":{"broker":"","port":1883,"clientId":"","user":"","pwd":"","topic":"inverter","intvl":0},"led":{"0":255,"1":255,"2":255,"act_high":false,"lum":255},"plugin":{"disp":{"type":0,"pwrSafe":false,"screenSaver":0,"graph_ratio":0,"graph_size":2,"rotation":0,"contrast":140,"data":255,"clock":255,"cs":255,"reset":255,"busy":255,"dc":255,"pirPin":255},"cst_lnk":"","cst_lnk_txt":""},"inst":{"intvl":30,"en":false,"rstMidNight":true,"rstNotAvail":false,"rstComStop":false,"strtWthtTime":false,"rstMaxMidNight":false,"yldEff":1,"rdGrid":true,"iv":[{"en":true,"name":"BKK","sn":xxx,"freq":0,"pa":2,"dis":true,"yield":[0,0,0,0,0,0],"pwr":[380,380,0,0,0,0],"chName":["1","2","","","",""]}]}}

### Error description

1) setup the ahoy-dtu and everything is working

2) I want to create a status page with some components in my home network and therefor I want to check some stuff from different components. For this I use curl to get some information.

3) when curl to the url of ahoy-dtu I normaly receive the page, but at ahoy-dtu I receive a warning for binary output.
curl ahoy-dtu.fritz.box => warning
curl --output - ahoy-dtu.fritz.box => binary data I have no idea what it is

4) when I just want to receive the http status code I leads to a redirection
curl --head ahoy-dtu.fritz.box => 302
and then if I use
curl --location --head ahoy-dtu.fritz.box
I receive an endles redirection till curl give up

On Tasmota or other home network clients it is working. Is this not possible with ahoy-dtu? Or how to make it possible?

knickohr, 2024-05-10:
Try curl on available REST-API endpoints :
{"avail_endpoints":{"inverter/list":"http://192.168.100.86/api/inverter/list","inverter/id/0":"http://192.168.100.86/api/inverter/id/0","inverter/alarm/0":"http://192.168.100.86/api/inverter/alarm/0","inverter/version/0":"http://192.168.100.86/api/inverter/version/0","generic":"http://192.168.100.86/api/generic","index":"http://192.168.100.86/api/index","setup":"http://192.168.100.86/api/setup","setup/networks":"http://192.168.100.86/api/setup/networks","setup/getip":"http://192.168.100.86/api/setup/getip","system":"http://192.168.100.86/api/system","live":"http://192.168.100.86/api/live","powerHistory":"http://192.168.100.86/api/powerHistory","powerHistoryDay":"http://192.168.100.86/api/powerHistoryDay","yieldDayHistory":"http://192.168.100.86/api/yieldDayHistory"}}

MaDill, 2024-05-10:
Thank you. Now I'm one step ahead. But I still receive the redirection loop till curl give up after 50 redirections.

First it use http://ahoy-dtu.fritz.box and then it get redirected to /wizard all till curl stops.

best regards

here are the verbose outputs from curl:

```
curl --compressed --location --head -v http://ahoy-dtu.fritz.box
* processing: http://ahoy-dtu.fritz.box
*   Trying 192.168.1.123:80...
* Connected to ahoy-dtu.fritz.box (192.168.1.123) port 80
> HEAD / HTTP/1.1
> Host: ahoy-dtu.fritz.box
> User-Agent: curl/8.2.1
> Accept: */*
> Accept-Encoding: deflate, gzip, br, zstd
> 
< HTTP/1.1 302 Found
HTTP/1.1 302 Found
< Content-Length: 0
Content-Length: 0
< Connection: close
Connection: close
< Location: /wizard
Location: /wizard
< Accept-Ranges: none
Accept-Ranges: none

< 
* Closing connection
* Issue another request to this URL: 'http://ahoy-dtu.fritz.box/wizard'
* Hostname ahoy-dtu.fritz.box was found in DNS cache
*   Trying 192.168.1.123:80...
* Connected to ahoy-dtu.fritz.box (192.168.1.123) port 80
> HEAD /wizard HTTP/1.1
> Host: ahoy-dtu.fritz.box
> User-Agent: curl/8.2.1
> Accept: */*
> Accept-Encoding: deflate, gzip, br, zstd
> 
< HTTP/1.1 302 Found
HTTP/1.1 302 Found
< Content-Length: 0
Content-Length: 0
< Connection: close
Connection: close
< Location: /wizard
Location: /wizard
< Accept-Ranges: none
Accept-Ranges: none

< 
* Closing connection
* Issue another request to this URL: 'http://ahoy-dtu.fritz.box/wizard'
* Hostname ahoy-dtu.fritz.box was found in DNS cache
*   Trying 192.168.1.123:80...
* Connected to ahoy-dtu.fritz.box (192.168.1.123) port 80
> HEAD /wizard HTTP/1.1
> Host: ahoy-dtu.fritz.box
> User-Agent: curl/8.2.1
> Accept: */*
> Accept-Encoding: deflate, gzip, br, zstd
> 
< HTTP/1.1 302 Found
HTTP/1.1 302 Found
< Content-Length: 0
Content-Length: 0
< Connection: close
Connection: close
< Location: /wizard
Location: /wizard
< Accept-Ranges: none
Accept-Ranges: none
```

knickohr, 2024-05-10:
Ahhh, OK, Update to a newer DEV-Version, this is a known Bug with the wizard.

MaDill, 2024-05-11:
OK. Thank you for the information. When this issue is known, then the bug can be closed, or?

MaDill, 2024-05-13:
I didn't test the dev version, because except this my setup is stable.

closed 2024-05-13 by MaDill