lumapu/ahoy #1611: <TOPIC>/<INVERTER_NAME_FROM_SETUP>/ch0/PowerLimit

andewo, 2024-05-10:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.8.83
Github Hash: 5ebfe5a

Hallo in die Runde,

kann mir jemand sagen ob dieses Topic "PowerLimit" noch zur Verfügung steht. Bekomme es nicht in HA geladen. Habe mir MQTT Explorer geladen - unter ch0 gibt es noch 12 topics - PowerLimit ist nicht dabei. Würde das gerne in HA für die Nulleinspeisung verwenden - ggf. muss ich das anders lösen....habe auf die aktuelle Ahoy-version upgedated...

andewo, 2024-05-11:
Hallo Lumapu, vielen Dank für die Rückinfo. Leider habe ich mich nicht eindeutig ausgedrückt. Kann man den Wert PowerLimit auch auslesen ? Ich kann den Wert mit <TOPIC>/ctrl/limit/0 setzen - Danke hierfür. Ich möchte danach aber auch den Wert auslesen können - so wie er in der Ahoy Übersicht angezeigt wird. 100% , 80% etc.

blueline13, 2024-05-11:
@andewo 
Du findest den aktuellen geschrieben wert im folgendem Feld: <DTU>/<Inverter>/ack_pwr_limit 

Den Topic kannst du als Acknowledge betrachten das der WR das Limit akzeptiert. Du solltest aber eine aktuelle Developer Version benutzen. Früher wurde da nur der Zeitstempel aktualisiert. Ich kann leider nicht genau sagen ab wann das geändert wurde.

blueline13, 2024-05-12:
@andewo 
Deswegen sagte ich ja eine aktuelle Developer. Die 0.8.118 unterstützt das, deine Version hat das noch nicht. Sieht dann so aus:

[image]

andewo, 2024-05-12:
@blueline13 - ok - an eine Developer-Version habe ich mich bis dato noch nicht ran getraut. Das werde ich mir mal anschauen. Wäre natürlich schön wenn das auch in eine aktuelle stabile Version Einzug hält...Nochmals vielen Dank für den Hinweis und den Beispiel - Screen-shot.

closed 2024-08-16 by lumapu