lumapu/ahoy #1612: 0.7.36 FW perfect - 0.8.83 FW shows getNtpTime failed

Philladelph, 2024-05-10:
My last FW 0.7.36 is running perfect. Older FW versions also always perfect. Tanks !

But when I do an update to the latest 0.8.83 FW it shows no date.
The Serial/Control always shows "getNtpTime failed"
My FritzBox router is working as NTP Server.

I would like to get the latest and greatest running

FW: 230817_ahoy_0.7.36_ba218ed_esp8266.bin
FW: 240217_ahoy_0.8.83_5ebfe5a_esp8266-de.bin

## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: HMS-1600-4T
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)
I am happy with Ahoy-DTU. 

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

lumapu, 2024-05-10:
The it would be may of interest to test our latest development version: https://fw.ahoydtu.de/fw/dev/

Philladelph, 2024-05-19:
[image]
[image]
[image]
[image]

Now totally struggling
Now nothing works anymore, even if I downgrade the firmware to an older version
Are the pictures helpful to you or do you need more from me?

closed 2024-05-19 by Philladelph