lumapu/ahoy #1613: [Bug] homeassistant discovery "name" enthält ungültige Zeichen

O1LER, 2024-05-10:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

circuit board

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a8fb5bcc833b64a0dd083686031808cec8

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

{"version":11,"wifi":{"ssid":"IoT","pwd":"","ap_pwd":"esp_8266","hidd":false,"dev":"AHOY-DTU","adm":"","prot_mask":61,"dark":true,"reb":true,"region":0,"timezone":1,"ip":"","mask":"","dns1":"","dns2":"","gtwy":""},"nrf":{"cs":15,"ce":0,"irq":2,"sclk":0,"mosi":0,"miso":0,"en":true},"ntp":{"addr":"pool.ntp.org","port":123,"intvl":10080},"sun":{"lat":,"lon":"offs":0,"offsEve":0},"serial":{"show":true,"debug":false,"prv":true,"trc":false,"mqtt":false},"mqtt":{"broker":"192.168.0.174","port":1883,"clientId":"","user":"mqtt","pwd":"","topic":"garage","intvl":0},"led":{"0":255,"1":255,"2":255,"act_high":false,"lum":255},"plugin":{"disp":{"type":5,"pwrSafe":false,"screenSaver":1,"graph_ratio":0,"graph_size":2,"rotation":0,"contrast":60,"data":255,"clock":255,"cs":255,"reset":255,"busy":255,"dc":255,"pirPin":255},"cst_lnk":"","cst_lnk_txt":""},"inst":{"intvl":15,"en":false,"rstMidNight":false,"rstNotAvail":false,"rstComStop":false,"strtWthtTime":false,"rstMaxMidNight":false,"yldEff":0.949999988,"rdGrid":true,"iv":[{"en":true,"name":"garage","sn":112184799070,"freq":0,"pa":0,"dis":false,"yield":[0,0,0,0,0,0],"pwr":[300,0,0,0,0,0],"chName":["","","","","",""]}]}}

### Error description

homeassistant discovery "name" enthält für json files ungültige Zeichen:

```
{"name":"CH1_U_DC\u0000@�?|\f�?�I�?�I�?�I�?$a#@","stat_t":"garage/garage/ch1/U_DC","unit_of_meas":"V","uniq_id":"112184799070_ch1_U_DC","dev":{"name":"garage","ids":"112184799070","mdl":"garage","cu":"http://192.168.0.167","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"voltage","stat_cla":"measurement"}
```
```
{"name":"CH1_I_DC\u0000@�?�%@lP�?\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000�%@","stat_t":"garage/garage/ch1/I_DC","unit_of_meas":"A","uniq_id":"112184799070_ch1_I_DC","dev":{"name":"garage","ids":"112184799070","mdl":"garage","cu":"http://192.168.0.167","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"current","stat_cla":"measurement"}
```

[image]

Wenn der Name manuell in MQTT explorer korrigiert und versandt wird, funktioniert die Erkennung in homeassistant ohne Probleme
```
{"name":"CH1_I_DC","stat_t":"garage/garage/ch1/I_DC","unit_of_meas":"A","uniq_id":"112184799070_ch1_I_DC","dev":{"name":"garage","ids":"112184799070","mdl":"garage","cu":"http://192.168.0.167","mf":"Hoymiles"},"exp_aft":95,"dev_cla":"current","stat_cla":"measurement"}
```
Die eigentlichen MQTT Nachrichten von Ahoy sind in Ordnung, keine komischen Zeichen im Namen. Ahoy war ausserbetrieb über den Winter, vorherige Firmwareversion unbekannt. Bei wiederinbetriebnahme hatte ich auf die aktuellste Version gewechselt.

Firmware 0.8.83 nochmals geladen, ohne änderung. 

Vorherige Firmware 0.8.36 geladen, nun keine Verbindung mehr. Heisst nun wohl AhoyDTU ausbauen und direkt am PC neu laden

lumapu, 2024-05-10:
danke für den Report. Der Fehler wurde schon gelöst, aber leider ist der Fix noch nicht in ein neues Release eingeflossen. Lade dir doch bitte mindestens die 0.8.84 auf deine DTU um den Fehler zu beheben.
https://fw.ahoydtu.de/fw/dev/

lumapu, 2024-05-10:
PS.: wir haben ein Changelog - das ist leider etwas versteckt, zeigt aber sehr gut was geändert wurde;
https://github.com/lumapu/ahoy/blob/development03/src/CHANGES.md

O1LER, 2024-05-11:
Vielen Dank für die Rückmeldung

closed 2024-05-11 by O1LER