lumapu/ahoy #1618: Invertereinstellungen in V0.8.118

Gubi2023, 2024-05-12:
in den WR-Eistellungen der neuesten Version wurde eine 
[image]
Funktion geändert: aus reset max-Values at midnight (oder so ähnlich) 
wurde  inlude reset max-Values (Incl. Tippfehler...) siehe screenshot  
Ich  hatte diese  nur Haken nur  gesetzt, aber der bewirkt anscheinend nichts mehr(?)

Gubi2023, 2024-05-12:
wozu ist dieser Haken dann noch da? dann  kann man ihn  gleich weglassen...

lumapu, 2024-05-12:
genau wie @MetaChuh geschrieben hat, kann man damit auswählen, ob die max-Werte auch zurückgesetzt werden sollen oder nicht. Ich finde es sehr sinnvoll.
Im Prinzip hat sich unter der Haube nicht viel getan, die Bennung des Hakens war falsch.

Gubi2023, 2024-05-12:
mir ist es auch nur aufgefallen, weil erst mit der 0.8.118 die max-Werte nicht mehr gelöscht wurden, sonst wurden sie immer auch nur mit dem einen Haken gelöscht....

Gubi2023, 2024-05-12:
ich hätte noch eine Frage zu "Werte bei Sonnenaufgang/untergang zurücksetzen":  werden sie zu diesem Zeitpunkt gelöscht, oder 
zu Start bzw. Ende der Kommunikation? (weil andernfalls die Wert  verzögert bzw zu früh gelöscht würden)

lumapu, 2024-05-16:
Die Werte werden zum Zeitpunkt des errechneten Sonnenauf- bzw. untergangs zurückgesetzt. Es handelt sich genau um die Zeiten, die man unter `/system` sehen kann

closed 2024-05-17 by Gubi2023