lumapu/ahoy #1619: [Bug] Instable Webserver since 0.9.87

lille2000, 2024-05-12:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.97

### Github Hash

Tbd

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Eps8266
3 Inverter der HM- Serie
Display
Deutsch
Keine Nutzung MQTT, API

### Error description

Seit 0.8.98 (inkl 0.8.118) ist dieses Setup nicht mehr zu gebrauchen. Nach wenigen Wechsel in den Webseiten Menüs scheint der Webserver abzustürzen.  
0.8.97 hat das Problem auch aber viel weniger häufig sodass dies die letzte nutzbare Version ist. 

[image]
[image]

knickohr, 2024-05-12:
Tja, mit 3 WRs (womöglich sogar mehrkanalige) würde ich auf einen ESP32 wechseln 😉

Es wird halt einfach zu viel für den „kleinen“.

lille2000, 2024-05-13:
Es sind insgesamt 5 Kanäle.
Was ist den der Grund das die Leistung mit steigendender Kanalanzahl nicht ausreicht?  Die Abfrage mit den Invertern ist vielleicht nach 2s fertig, dann hat man bei einem 15 s Intervall noch 13s, in denen nichts passiert und die man frei hat.

Oder ist es mangelnder RAM, sodass der Speicher nicht ausreicht für eine Webseitenanfrage + Inverter-Routinen?

Ich habe mir nun  einen esp32 gekauft.

dtuuser, 2024-05-13:
Hatte am Anfang auch esp8266. Bin dann aber schnell auf esp32 umgestiegen. Läuft alles viel stabiler.

kkloesener, 2024-05-26:
Ich habe das Problem allerdings auch immer wieder.
Umgebung 1 Inverter HM, ESP32, kein MQTT oder API.

Ich beobachte das Problem primär dann, wenn das System längere Zeit(>24h) ohne reboot läuft. 
Ich habe einen Batteriespeicher von Zendure noch vor dem Inverter, so dass es regelmäßig dazu kommt, dass der Inverter bis morgens um 6 mit Strom versorgt ist und so die DTU weder neustartet, noch die Tageswerte zurücksetzt.

lumapu, 2024-05-26:
an alle mit ESP32, ihr könnt mir helfen die Fehler zu finden, wir habe ich hier beschrieben:
https://github.com/lumapu/ahoy/issues/1600#issuecomment-2131461052

lumapu, 2024-05-30:
bitte nochmal mit der nächsten dev testen, es gab diesbezüglich Verbesserungen (`0.8.123`)

lille2000, 2024-06-04:
Mit 0.8.123 ist es besser, es halt länger durch, aber nicht so, dass man diese Version länger verwenden möchte.
Sobald man per Browser zwischen den Unterseiten wechselt passiert es sehr bald, dass nur noch „statischer“ Text dargestellt wird. Alle Livedaten (Settings, Live, System, Update) fehlen als Menüpunkt ( siehe Bilder oben)
Zb statt 
„ Alle 15 Sekunden werden die Daten aktualisiert“ steht:
„ Alle Sekunden werden die Daten aktualisiert“

Mit 0.8.98 ist es weniger häufig, aber auch erlebbar.

ralfberger, 2024-06-07:
kann ich bestätigen, 0.8.123 und auch 0.8.124 läuft der Webserver instabil. 
Bei mir ist nur ein Wechselrichter aktiv und MQTT sendet alle 1800 s.
Ich werde dann auch mal 0.8.97 testen und langfristig auf ESP32 umsteigen.

juepi, 2024-06-12:
Kleines Update von meiner Seite: ich kann nicht beurteilen ob's mit diesem Bug zu tun hat, aber die 0.8.124 lief bei mir generell extrem instabil. Ständige reboots, im WebUI meldete er oft dass er mit dem NTP server nicht syncen kann und deswegen mit dem Inverter nicht kommuniziert.
Konnte auch die config teilweise nicht mehr speicher ("config save failed").
Habe jetzt auch auf 0.8.97 downgegraded, schaut so weit wieder stabil aus.

Muss dazusagen, dass ich seit ca. 1 Woche über FHEM auf Nulleinspeisung regle, das Powerlimit meines Inverters (1 HM-300 momentan) wird max. alle 5 Sekunden neu gesetzt, entsprechend auch das "Interval" Setting in der Inverter-config auf 5sec reduziert - eventuell Überlastung des ESP8266? 

lg,
Jürgen

lumapu, 2024-06-12:
es ist glaube ich mich zu erinnern ratsam zumindest die Intervalle unterschiedlich zu setzen, z.B. das eine auf 5, das andere auf 7 Sekunden. Insgesamt ist der ESP8266 aus meiner Sicht nicht (mehr) wirklich geeignet, da Ahoy inzwischen zu "schwer" geworden ist. Du könntest mal die minimal-Variante testen, aber auf die lange Sicht würde ich über einen ESP32 nachdenken.

juepi, 2024-06-13:
> es ist glaube ich mich zu erinnern ratsam zumindest die Intervalle unterschiedlich zu setzen, z.B. das eine auf 5, das andere auf 7 Sekunden. Insgesamt ist der ESP8266 aus meiner Sicht nicht (mehr) [...]

Danke Lukas, dann werd ich mich einfach gleich mal dran machen, mein Ahoy zumindest auf einen Lolin S2 Mini zu migrieren.

knickohr, 2024-06-13:
Warum wechselt ihr denn nicht gleich auf einen ESP32-S3 N8 mit mind. 8MB Flash ?  Es wird in Zukunft Funktionen in Ahoy geben die nicht mehr auf die 4MB passen.

knickohr, 2024-06-13:
Das ist OK, nur wird damit die neue Funktion „bedarfsoptimierte Leistungsregelung“ die in Kürze kommt, nicht funktionieren 😉

juepi, 2024-06-13:
Oha, na dann werd ich mir wohl mal doch eher früher als später einen bestellen müssen 😁

lille2000, 2024-06-22:
Ich habe nun auf einen ESP32 WROOM D1 Mini  (wegen der Pin Kompatibilität ) gewechselt. 
Damit läuft alles geschmeidig und sehr stabil. So bin war das gar nicht mehr gewohnt vom ESP8266.

Auch die Funkstatistik ist nochmal besser (wobei die bisher auch schon sehr gut war).

stefan123t, 2024-11-02:
@lille2000 welche Version hast Du aktuell drauf ? Die Version master 0.8.144 oder development03 0.8.152 ?
Können wir das Issue schließen oder tritt das Problem noch weiter auf ?

knickohr, 2024-11-03:
Stable ist 0.8.140 !