lumapu/ahoy #1620: Max Power + Yield Today not resetted

elisenjens, 2024-05-12:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.118

### Github Hash

85e3dea

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard

### Debug Serial Log output

```shell
no log
```

### Error description

Like some versions before: Max Power + Yield Today are not resetted even though "Reset values and YieldDay at midnight" and "Inlude reset max values" are flagged.

Starfoxfs, 2024-05-13:
Ich hab hier ein sagen wir mal ähnliches Problem, ich hab die 0.8.83 installiert und seitdem extrem hohe Tageswerte. Hab ich hier was falsch gesetzt in den Einstellungen des Wechselrichter (siehe Bild unten) ?

[image]
[image]

ralfberger, 2024-05-15:
@Starfoxfs 
Hast du die entsprechenden Optionen jetzt gesetzt und werden die Max Werte gelöscht?

Starfoxfs, 2024-05-15:
Ja hab ich, und in der 0.8.83 werden die um Mitternacht gelöscht. Nur die Beschreibung ist etwas irreführend, weil ein Gesamtertrag erwähnt wird und es wohl ein Übersetzungsfehler ist. 
Siehe auch hier:
https://github.com/lumapu/ahoy/issues/1609#event-12817470319

lumapu, 2024-05-15:
danke für den Hinweis mit dem WLAN, das wird es wohl sein

elisenjens, 2024-05-15:
hat bei mir in den älteren Devs aber funktioniert; hab nichts geändert an meiner Konfig

elisenjens, 2024-05-19:
working again in 0.8.119 THX

closed 2024-05-19 by elisenjens