lumapu/ahoy #1621: Shutdown Problem

slg60, 2024-05-12:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Habe seit gestern Mittag das Problem, dass sich die Ahoy komplett weg hängt. Display ist aus und ich kann sie nicht mehr erreichen, mqtt Nachrichten kommen auch keine mehr. Ein Power cycle behebt das Problem vorübergehend, bis zum nächsten Ausfall. Die Ahoy lief vorher mit der Version 08.101 Wochenlang ohne Probleme. Seit gestern dann diese Ausfälle. Zu beobachten am Tage wenn ordentlich produziert wird. Nachts läuft sie durch.

Maßnahmen bisher:
220u Kondensator an die 3,3V-> Keine Verbesserung
Update auf die 08.118-> Keine Verbesserung
Netzteil Tausch-> Ergebnisse kann ich erst morgen berichten. (Aktuell 5V 1A Netzteil)

Was mich wundert ist, dass die Ahoy nicht wieder startet. Sie bleibt einfach aus. Bei einem Problem mit der Stromversorgung hätte ich erwartet das sie neu startet, aber nicht das sie ganz aus bleibt. Hoffe nicht das es meinen ESP erwischt hat. Der ist erst 3 Monate alt...
Ideen?

slg60, 2024-05-13:
Auch der Netzteil/Kabel Tausch hat keine Besserung gebracht. Eben hat die Ahoy wieder ihren Dienst eingestellt...

kiu77, 2024-05-13:
Flashe das Teil mal komplett mit Erase des Flash-Speichers.

Solche schrägen Dinge sind mir auch schon untergekommen. Letztlich hat das Flashen immer irgendwie geholfen.

slg60, 2024-05-13:
Hmm, warum dann auf ein mal... Ich habe bis auf das Inverter Intervall seit Wochen nichts geändert. Und auch mit dem 3 Sekunden Intervall läuft das Teil seit bestimmt 2 Wochen ohne Probleme. Und jetzt auf ein mal... Schon komisch....

kiu77, 2024-05-13:
Kann ich Dir auch nicht sagen. Außer dass manchmal digitale Elektronik scheinbar Sehnsucht danach hat, sich endlich mal analog auszuleben. Falls Du ein Multimeter hast, kannst Du auch mal die %V und 3,3V nachmessen. Oszilloskop wäre noch besser, weil man da Schwankungen sehen kann bzw. beurteilen, ob die Versorgung "weich" ist.

slg60, 2024-05-13:
Ist ein esp32 wroom mit cm2300 und Display, verbunden über steckbare Kabel. Spannungsversorgung habe ich ja nun schon gewechselt. Auch mit dem 2A Netzteil steigt das aus.

slg60, 2024-05-14:
Ich habe gestern nochmal Messungen durchgeführt mit beiden Netzteilen. Die 5V lag immer deutlich über 5V (5,08-5,13V, die 3,3 lag aber teilweise minimal unter 3,3V (3,28V) und das gemessen mit einem Multimeter. Ich bin jetzt auf ein Netzteil gegangen welches 5,3V liefert und jetzt liegt die 3,3V auch immer im grünen Bereich. Bisher läuft die DTU seit heute Morgen ohne Absturz. Sollte es das gewesen sein, dann bin ich erschrocken wie empfindlich das Konstrukt auf die Spannungsversorgung reagiert.

slg60, 2024-05-15:
Bis heute um 17 Uhr lief die Dtu ohne probleme durch, dann ist sie wieder neu gestartet, dieses mal ist die aber wenigstens von alleine wieder hoch gefahren. Irgendwas passt da auf jeden Fall nicht.

lumapu, 2024-05-30:
bitte die aktuelle Dev testen, es wurde einiges im Bezug auf Stabilität verbessert (`0.8.123`)

slg60, 2024-06-29:
Getestet und Problem nach wie vor sporadisch vorhanden. Heute blieb das Display mal an aber die Ahoy war über Netzwerk nicht mehr erreichbar, obwohl sie in der Fritzbox noch registriert war. Reboot tat gut. Auffällig ist das das immer in der Phase passiert wenn der Himmel wolkenlos ist und Volle Leistung produziert wird. Ist schwer bei zu kommen. Echt nervig das Problem. Gibt es noch Ideen was ich noch machen kann?

Gubi2023, 2024-06-29:
siehe auch #1651

slg60, 2024-06-29:
> siehe auch #1651

Sieht aber nach einem anderen Problem aus. Hatte bis gestern die 123 drauf, die lief bis gestern wochenlang ohne ein einziges Problem und auf ein mal wieder aus heiterem Himmel shutdown ohne reboot. Habe dann auf die 127 geupdated, aber das Problem scheint nicht gefixt, irgendwas löst da ein Problem aus. Hatte schon drüber nachgedacht ob es evtl etwas mit dem WLAN Kanal 13 zu tun hat. Meine Fritzbox wechselt häufiger den Kanal zwischen 6 und 13. Kann mich erinnern das es mit Kanal 13 mal Probleme mit dem 8266 gab...

Gubi2023, 2024-06-29:
mir fiel nur die Analogie auf: 
0.8.123 schon nach 6 St. nicht mehr erreichbar, auch per ping nicht, obwohl sie in der Fritzbox noch angezeigt wird/
Heute blieb das Display mal an aber die Ahoy war über Netzwerk nicht mehr erreichbar, obwohl sie in der Fritzbox noch registriert war.

slg60, 2024-06-29:
Das wiederum deckt sich mit meiner Erfahrung mit der 127.

stefan123t, 2024-11-03:
@slg60 kannst Du es bitte mit der aktuellen 0.8.144 release oder 0.8.152 development03 noch einmal prüfen ob das Problem noch auftritt ? Danke im Voraus!

slg60, 2024-11-05:
Moin, ehrlich gesagt fahre ich aktuell mit der 0.8.127 sehr stabil ohne reboots. Ich hadere mit mir jetzt auf die 152 zu wechseln. Ich verfolge das Thema parallel und es scheint ja viele zu geben die auch mit der 152 noch in diese Problematik laufen. Hierzu gibt es coredups etc.