lumapu/ahoy #1622: Webserial Timestamp 2 Stunden vor Uhrzeit

slg60, 2024-05-13:
Mir ist eben aufgefallen, das in der Webserial der Timestamp 2 Stunden vor der vom NTP bereitgestellten Zeit steht. Kann ich den irgendwo separat einstellen oder ist das ein Bug? Aktuelle Beta 118.

Gubi2023, 2024-05-13:
Das ist leider so, wenn die DTU sich neu startet und das  Webserial geöffnet war, ist die Zeit im Webserial falsch. Durch aktualisieren wird sie wieder berichtigt. Der Fehler ist seit langem so vorhanden, warum weiss ich auch nicht...

slg60, 2024-05-13:
Was meinst du mit aktualisieren? Bei mir liegt die Zeit immer konstant 2 Stunden hinter der ntp Zeit. Die wird ja erfolgreich von der Dtu bezogen. Nur offenbar nicht für die Webserial herangezogen.

Gubi2023, 2024-05-13:
einfach webserial neu starten oder m strg+R drücken

slg60, 2024-05-13:
Bin auf dem Smartphone unterwegs, Neustart oder aktualisieren verändert bei mir nichts an der Zeit... Immer 2 Stunden Offset.

Gubi2023, 2024-05-13:
tja, dann weiss ich auch nicht weiter....

Peertje-62, 2024-05-13:
Settings,
[image]

Peertje-62, 2024-05-14:
Hi, set settings; system; config; time zone?
[image]

slg60, 2024-05-14:
GMT+1 is set as on your screenshot. For Summertime it may be +2 but then it would still be wrong. Why it is not syncing to NTP?

Peertje-62, 2024-05-14:
Did you set in Settings Sunrise & Sunset, Latitude; Longitude aswell ?

slg60, 2024-05-14:
Yes

Peertje-62, 2024-05-14:
Hi, Yes 
- You have set it or 
- Yes its working correct now
I am have no more options ( I am running development version 0.8.117) 
kind regards

slg60, 2024-05-14:
I've set this to my location. Im using an 30min offset also, but this should not affect the time...

rmayergfx, 2024-05-15:
@slg60 
Die Uhrzeit im Webinterface auf dem Smartphone kommt nicht von AhoyDTU sondern vom Smartphone.  Ist mir auch schon aufgefallen, wenn ich mich remote per VPN auf das Device verbunden habe und in einer anderen Zeitzone war. 
Prüfen ob auf dem Smartphone die Uhrzeiten richtig eingestellt sind, ggfs mal auf manuell umschalten und dann erneut testen.

slg60, 2024-05-15:
Hmm, ist auf Smartphone und PC identisch und die Zeiten sind dort auch richtig eingestellt. Evtl liegt's am Browser. Ich schau mal.

stefan123t, 2024-11-03:
@slg60 mit WebSerial Timestamp meinst Du die Timestamps unter WebSerial die vor jeder Logmeldung stehen.
Die sind bei mir korrekt dargestellt:

[image]

Settings > System Config
Timezone: +1

Settings > NTP Server
NTP Server / IP: pool.ntp.org
System Time: 2024-11-03, 22:53:51

Also der Timestamp hier passt mit dem Timestamp unter Webserial.