lumapu/ahoy #1623: Unable to write letter of Serial

Maeckes89, 2024-05-14:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: Vroom 32
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.118
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

The Serial of my Hoymiles HMS-1600 is something like ####A###F#CF (# is a number)

But I'm unable to write a letter in the Serial-field in the inverter setup. So - what to do?

lumapu, 2024-05-16:
are you sure that you have a HMS-1600 or could it be a HMS-1600W

lumapu, 2024-05-30:
@Maeckes89 are there any updates?

LnBsN, 2024-06-20:
Bonjour,

Je me retrouve avec le même problème, j'ai un micro onduleur HMS-1000-2T comportant 2 lettres dans le numéro de série, sauf que je ne peux écrire que des chiffres.
Auriez vous une solution pour écrire le numéro de série avec chiffres et lettres ?

Il s'agit de : 
La version 0.8.83
ESP 32 
Radio CMT2300A

Cordialement

lumapu, 2024-06-21:
you can use the development version: https;//fw.ahoydtu.de/fw/dev/

LnBsN, 2024-06-23:
> Vous pouvez utiliser la version de développement :
> 
> https://fw.ahoydtu.de/fw/dev/

Bonjour,

Cela a fonctionné avec la version .127
Merci pour votre aide !

Cordialement

Leitzbach1, 2024-07-09:
I had the same issue, i just changed the html input field type="number" to type:"text" and could enter letters.

closed 2024-08-16 by lumapu