lumapu/ahoy #1624: no scale in course/history-view [Bug]

Windsurfer87, 2024-05-14:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

[image]

### Debug Serial Log output

```shell
0.8.83 - 5ebfe5a
---------------
webSerial, buffer overflow!
Hoymiles-HM1500/ch0/ALARM_MES_ID: 54.000 
15:46:44.128 I: Hoymiles-HM1500/ch0/YieldDay: 5279.000 Wh
15:46:44.132 I: Hoymiles-HM1500/ch0/YieldTotal: 2013.583 kWh
15:46:44.136 I: Hoymiles-HM1500/ch0/P_DC: 1118.000 W
15:46:44.140 I: Hoymiles-HM1500/ch0/Efficiency: 94.991 %
15:46:44.144 I: Hoymiles-HM1500/ch0/MaxPower: 1224.400 W
15:46:44.147 I: 
15:46:44.152 I: MQTT disconnected, reason: TCP disconnect
15:46:44.395 I: (#0) RX  27ms | 27 CH75 | 95 01
15:46:44.396 I: (#0) RX 147ms | 27 CH40 | 95 03
15:46:44.397 I: (#0) RX 198ms | 27 CH40 | 95 84
15:46:44.740 I: (#0) RX  44ms | 27 CH61 | 95 02
15:46:44.743 -----
15:46:44.743 I: com loop duration: 595ms
15:46:44.744 -----
:46:44.040 I: Hoymiles-HM1500/ch2/Irradiation: 68.775 %
15:46:44.044 I: Hoymiles-HM1500/ch2/MaxPower: 317.200 W
15:46:44.048 I: Hoymiles-HM1500/ch3/U_DC: 31.500 V
15:46:44.051 I: Hoymiles-HM1500/ch3/I_DC: 8.880 A
15:46:44.054 I: Hoymiles-HM1500/ch3/P_DC: 278.900 W
15:46:44.058 I: Hoymiles-HM1500/ch3/YieldDay: 1326.000 Wh
15:46:44.062 I: Hoymiles-HM1500/ch3/YieldTotal: 502.553 kWh
15:46:44.066 I: Hoymiles-HM1500/ch3/Irradiation: 69.725 %
15:46:44.070 I: Hoymiles-HM1500/ch3/MaxPower: 326.500 W
15:46:44.074 I: Hoymiles-HM1500/ch4/U_DC: 31.500 V
15:46:44.077 I: Hoymiles-HM1500/ch4/I_DC: 8.900 A
15:46:44.080 I: Hoymiles-HM1500/ch4/P_DC: 279.600 W
15:46:44.084 I: Hoymiles-HM1500/ch4/YieldDay: 1314.000 Wh
15:46:44.088 I: Hoymiles-HM1500/ch4/YieldTotal: 505.540 kWh
15:46:44.092 I: Hoymiles-HM1500/ch4/Irradiation: 69.900 %
15:46:44.096 I: Hoymiles-HM1500/ch4/MaxPower: 322.200 W
15:46:44.100 I: Hoymiles-HM1500/ch0/U_AC: 249.600 V
15:46:44.103 I: Hoymiles-HM1500/ch0/I_AC: 4.260 A
15:46:44.107 I: Hoymiles-HM1500/ch0/P_AC: 1062.000 W
15:46:44.111 I: Hoymiles-HM1500/ch0/Q_AC: 24.300 var
15:46:44.114 I: Hoymiles-HM1500/ch0/F_AC: 50.010 Hz
15:46:44.117 I: Hoymiles-HM1500/ch0/PF_AC: 1.000 
15:46:44.120 I: Hoymiles-HM1500/ch0/Temp: 57.700 °C
```

### Error description

The history/course has no scale:

[image]

Gubi2023, 2024-05-14:
Skalierung gibt es nur in den Dev-Versionen. Siehe https://fw.ahoydtu.de/fw/ Am besten gleich die aktuelle nehmen, läuft relativ stabil

Windsurfer87, 2024-05-14:
Das hatte ich schon getestet - zumindest die DE-Version. Und hier fehlt mir der Verlauf-Tab komplett:

[image]

Windsurfer87, 2024-05-14:
Manueller Aufruf der Seite sieht so aus:

[image]

Gubi2023, 2024-05-14:
hast du die 240510_ahoy_0.8.118_85e3dea_esp8266-all-de.bin  verwendet? ausserdem müssen bei den Einstellungen Sonnenaufgang und -Untergang Koordinaten eingetragen sein

Windsurfer87, 2024-05-14:
Nein, ich habe die ohne "all" verwendet. Mit der genannten sieht es besser aus - danke für den Tipp!

Loetnase, 2024-05-16:
Wenn jetzt alles i.O. ist darfst du das Issue gerne schliesen :-)

Windsurfer87, 2024-05-16:
Ich hatte mit der "all" Version noch ein anderes Problem, was ich aber noch nicht genau eingrenzen kann. Aber das Skalen-Problem ist gelöst - Danke.