lumapu/ahoy #1627: A second Hoymiles Inverter can not be added

Maddiehn22, 2024-05-17:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.83

### Github Hash

5ebfe5a

### Build & Flash Method

ESP Tools (flash)

### Setup

Add a MQTT server with 5 seconds respond time

### Error description

Another Hoymiles Inverter with serial number above 138999999999 (in my case (141xxxxxxxxx617)) can not be added
and marked red in the serial number entry field 
[image]

Ollipop030, 2024-05-17:
Shouldn't the serial number be 12 characters long?

Maddiehn22, 2024-05-17:
12 characters are given on the labele oft the inverter.

Maddiehn22, 2024-05-17:
The shown screenshot is with typo :-(

Ollipop030, 2024-05-17:
What inverter do you have?

Maddiehn22, 2024-05-17:
Hoymiles HM-800

knickohr, 2024-05-18:
Is it really a HM not a HMS or HMT or something with H….-W…. ?

Maddiehn22, 2024-05-20:
Dear all,
disregard! My neighbour sent me the type plate of his inverter.
It is NO HM800 but a  HMS-800-2T WiFi.
Sorry for disturbance.

Maddiehn22, 2024-05-20:
The inverter is NO HM800 but a  HMS-800-2T WiFi.
Sorry for disturbance.