lumapu/ahoy #1628: [Bug] HMT Anzahl der Channels

geronet1, 2024-05-17:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.93 ETH

### Github Hash

12345

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

HMT-1800

### Error description

Wenn man HMT-1800 mit 4 Anschlüssen hat zeigt er 6 Kanäle bei Live an, die letzten zwei zeigen alle "0".
Man kann im Setup nur 4 Kanäle einstellen (korrekterweise) aber den 5. und 6. Kanal mit der Max_Module_power nicht auf Null stellen da Standard 400 W ist. Nur in der exportieren Konfigdatei kann man das ändern und somit die zwei nutzlosen Kanäle ausblenden lassen.

Über MQTT sendet er auch die zwei nicht vorhandenen Kanäle raus.

lumapu, 2024-05-17:
hast du den Anfang der Seriennumer für mich? Die ersten 4 Stellen reichen aus.

geronet1, 2024-05-18:
1361

Kannst du mir mal die Symptomatik hinter 

setup.html Zeile 842:
`sn = Math.floor(sn / Math.pow(2, 32));

                        var max = 1;
                        switch(sn & 0x00f0) {
                            case 0x0010: max = 1; break;
                            case 0x0040: max = 2; break;
                            case 0x0060: max = 4; break;
                            case 0x0080: max = 6; break;
                        }`
erklären? Warum 2^32?
Im Prinzip mit der Berechnung die wirklich vorhandenen Kanäle feststellen und Max_Module_Power bei den nicht benutzten auf 0 setzen. Bei der MQTT-Ausgabe dann entsprechend prüfen.

lumapu, 2024-05-18:
Ganz kurz: die ersten 4 Stellen werden extrahiert

klar zuerst mal bisschen Javascript:
```javascript
let ser = 116411223344
let sn = parseInt(ser, 16);
sn = Math.floor(sn / Math.pow(2, 32));
console.log("0x" + sn.toString(16))
```
Output:
`> "0x1164"`

Die Seriennummer wird in Hex angegeben. Zuerst wird diese in dez gewandelt und dann durch 2^32 geteilt (in Hex `0x100000000`), also einer Zahl, die die letzen 8 Stellen "löscht". Mit `floor` wird immer abgerundet. Das Ergebnis sind die ersten 4 Stellen der Seriennummer im dezimalen Zahlenraum.

lumapu, 2024-05-18:
Ich kann übrigens den Fehler (noch) nicht nachvollziehen:

<details>
<summary>Inverter settings 1</summary>

[image]
</details>
<details>
<summary>Inverter settings 2</summary>

[image]
</details>

lumapu, 2024-05-18:
Problem verstanden und behoben 😊. Die interpretation innerhalb von Ahoy war falsch. Update kommt jetzt dann gleich 0.8.120

geronet1, 2024-05-19:
Moment, so einfach ist es nicht ;-)
Jetzt weist du dem HMT-1800-4T die Struktur vom "hms4chAssignment" zu, in der gibt es aber nicht die Datenpunkte vom Drehstrom-Inverter wie FLD_UAC_1N oder FLD_UAC_31..
Abhilfe wäre eine ne extra hmt4chAssignment zu erstellen. Oder den "Umweg" über Max_Module_power verwenden wie ich oben beschreiben habe.
Danke für die Erklärung übrigens :-)

geronet1, 2024-05-19:
Achso, bei Inverter-Info steht kein passender Typ:

Model | HMT-0

geronet1, 2024-05-19:
Außerdem ist noch ein Schreibfehler in hms_Defines.h Zeile 31:

> const char* const fields[] = {"U_DC", "I_DC", "P_DC", "YieldDay", "YieldWeek", "YieldTotal",
        "U_AC", "U_AC_1N", "U_AC_2N", "U_AC_3N", "UAC_12", "UAC_23", "UAC_31", "I_AC",
        **"IAC_1", "I_AC_2", "I_AC_3",** "P_AC", "F_AC"....

Soll wohl "I_AC_1" heißen.
"UAC_12", "UAC_23", "UAC_31" sollten wohl dann konsequenterweise auch so aussehen:
"U_AC_12", "U_AC_23", "U_AC_31"

lumapu, 2024-05-19:
danke für die Hinweise, schade einen Versuch war es wert. 
Ja dann brauchen wir ein neues Assignment. Ist das ein exotischer Wechselrichter? Mir war dieser nie aufgefallen.

geronet1, 2024-05-19:
Die Serie gibts schon länger
Siehe https://www.shinetech-power.de/wp-content/uploads/2023/10/datasheet_hmt-160018002000_emeaapac_de_v202303.pdf

Soll ich dir einen PR erstellen?
Schaust du auch mal den anderen PR mit der JSON  Ausgabe an? Die max-Temperatur haste da ja schon extrahiert 😁

lumapu, 2024-05-20:
kannst du eine rohe Payload posten oder wie du gesagt hast einen PR - wie es dir lieber ist.
Ja deinen PR habe ich schon im Auge - eigentlich würde ich gerne davor noch relaesen - der ESP32 macht mir aber noch zu viele Sorgen bzgl. Reboots

geronet1, 2024-05-21:
> kannst du eine rohe Payload posten
wie mach ich das?

geronet1, 2024-05-21:
Schon fertig, evtl. muss man noch den Wert von HMT4CH_PAYLOAD_LEN, das kann ich nicht rausfinden.
[07dfb5f](https://github.com/lumapu/ahoy/pull/1541/commits/07dfb5fcd3a27afb20c4b4554d0c53af91a095b2)

lumapu, 2024-05-22:
die `HMT4CH_PAYLOAD_LEN` kannst du aus dem Webserial lesen, hier beispielhaft vom HMS2000-4T:
`12:41:59.062 I: (#2) Payload (66)` also die Länge beträgt `66`

geronet1, 2024-05-25:
Passt, Wert hat schon gestimmt und ist der gleiche wie beim 6Ch.

lumapu, 2024-05-25:
ok, danke für die Bestätigung

closed 2024-06-10 by geronet1