lumapu/ahoy #1629: [Bug] HMS-2000 Power Factor + Power Factor wird falsch ausgewertet

DanielR92, 2024-05-18:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

board

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.1030015

### Github Hash

dda2a451

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

[image]

### Error description

Die Daten werden nicht korrekt entnommen. 

Model | HMS-2000 (Serial: 11648072)
Firmware Version / Build 1.00.14 (build: 2021-12-30 11:23)
Hardware Version / Build 2.56 (build: 7/2022)
Hardware Number 10127100
Bootloader Version 0.01

lumapu, 2024-05-18:
Das liegt an der Firmwareversion und ist bekannt, ich weiß nicht ob wir sogar schon ein Issue dazu haben.

stefan123t, 2024-11-03:
@lumapu zumindest ist mir bisher bei meiner Aktion keines aufgefallen :wink: 

Ich sehe hier, dass der Power Factor mit 0 ausgegeben wird und Reactive Power -0,1 var ist.

@DanielR92 was ist denn hier genau verkehrt bzw. hast Du einen Log Mitschnitt der HMS Pakete vom WR zur AhoyDTU, damit wir evtl. damit das Problem besser nachvollziehen können ?