lumapu/ahoy #1631: HMS

ogowi, 2024-05-18:
0C 04 20 00 00 0B 08 FC 07 30 00 0F 09 F9 00 01 02 3F 00 32 0A 55 00 08 09 F9 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 03 02 58 09 CB 07 A3 13 92 12 8E 40 00 07 D0 00 10 50 00 00 01 13 9C 01 90 00 32 60 04 00 01 09 E2 0A 10 08 7E 80 00 00 00 08 44 01 2C 08 A0 09 6F 09 B4 01 2C 90 00 00 00 00 5F B0 00 00 00 01 F4 00 5F 70 02 00 01 27 10 A0 02 00 00 00 00

rmayergfx, 2024-05-18:
????

lumapu, 2024-05-18:
Can you provide more infos? I think you have pasted a unknown grid profile. 
Sadly you didn't provided the Ahoy version you're running at. In the development version it is already included. The profile name is `AT_TOR_Erzeuger_default`.

Can you please verify this and give feedback?

closed 2024-05-19 by ogowi