lumapu/ahoy #1632: Feature Request : Automatic Max, output in Watts

BenjiKimba, 2024-05-18:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

I have a HMS 1400 t4 with 2 solar panels on it delivering 800W max.
Now I want to add a third panel for more light sensitivity but do not want more output dan 800W.
Is it possible to modify the Active Power Control to do this automatically at his own ?
Its not the same as zero export, no need for data outside, only watching at AC Power and if the value go above 800W
than pull the output lower.
Thanks in advance

rmayergfx, 2024-05-18:
You don´t need any automation for this. HMS-1600 can be set from 600 to max with AhoyDTU, so if you set the limit in "W" it will always the output power is always reduced to the desired value, regardless of what is currently present. If you are below the output power, nothing happens at first; only when the value is above it is it reduced. This is done internally by the inverter; this automatic function is built in at the factory.

BenjiKimba, 2024-05-18:
Yes, I forgot to tell that the third panel is not in an optimal place, only for a few ours, so if I set the output
manually at 800W ( 200W per panel I Think ), then I have at max. 600W, min 400W ?

rmayergfx, 2024-05-18:
No, you didn´t understood, read the manual for your inverter. you are reducing the output to a max. of 800W. If this is not reached, it will not reduce anything.

BenjiKimba, 2024-05-18:
I think only if you have all 4 panels on it.

knickohr, 2024-05-18:
If you want max power 800W

- you have to set the inverter to 800W for 4 panels
-  to 1067W for 3 panels
- to 1600W for 2 panels /unlimited)

BenjiKimba, 2024-05-18:
That's true, but the 3th panel is not in optimal position, also one of the 2 panels became shading in Winter,
with my suggestion you can optimize the Yield dynamically  for this.

knickohr, 2024-05-18:
This is not possible until now.

A future Version of Ahoy will have an „demand optimized power control“.

Then you can set the SetPoint to -800W an your panels and inverter is able to deliver max. 800W to the grid.

But it’s still a beta version  running on my DTU.

For more information, please join the Discord https://discord.com/channels/984173303147155506/1211365440253726851

BenjiKimba, 2024-05-18:
Thank you for your effort on this.
I have only some Arduino experience and no idea about MqTT and all that stuff,
but I think its also with that possible, the same way how to make a zero export, but then 
reading the value of Active Power Control from the ESP ?

BenjiKimba, 2024-05-23:
@knickohr 
No need for an extra button, only improved Active Power Control.
AhoyDTU knows  the capacity of the inverter and the panels add to it written in Inverter/ Max Module Power.
So he can compensate the panels automatically even if you want to make a zero export .

By the way, no problem for me if someone want to answer me in German language.