lumapu/ahoy #1634: Active Power Control: Am nächsten Tag immer wieder auf 100 % HM-1500

Andy811963, 2024-05-19:
## Hardware
  * [ *] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname:  HM-1500
Retailer URL: ______

### nRF24L01+ Module
* [ *] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [* ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ *] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.119
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [* ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [*
[image]
[image]
[image]
 ] Setup settings (use our templates ... to be added)

Gubi2023, 2024-05-19:
Haken gesetzt??  
[image]

Andy811963, 2024-05-19:
Hallo zusammen,

Danke für die schnellen Antworten.

Aber ich finde den Haken nicht bei mir ?!

Gruß Andy
[image]

Gubi2023, 2024-05-19:
@MetaChuh   Ah! Er arbeitet noch mit V0.7.51! da kenne ich mich nicht mehr aus (zu alt)

Andy811963, 2024-05-19:
Hallo,

wie kann ich die Version erhöhen ?
Habe regelmäßig die Updates gemacht. Nun 08.120.
Sorry für die blöde frage, habe aber nur Halbwissen.

knickohr, 2024-05-19:
Wähle da mal aus persistant und prozent.

[image]

Andy811963, 2024-05-20:
Danke für die Tipps

closed 2024-05-20 by Andy811963