lumapu/ahoy #1635: Skalierung der Gesamtleistung falsch

Gubi2023, 2024-05-19:
bei den neueren Devs gibt es ein Problem mit der x-Achsen-Skalierung bei der Gesamtanzeige. Der linksseitige Inhalt wird abhängig vom Abfrageintervall gelöscht und die Anzeige nach rechts geschoben, sodass links eine Lücke entsteht. Bei mir wird dann pünktlich jede halbe Stunde die Skalierung neu kalibriert und das Spiel geht von vorne los. siehe Screenshot

[image]

Mein Abfrage-Intervall beträgt 15 sec. 
Bei 5-sekündigem Abfrage-Intervall scheint es aber zu funktionieren

strepwo, 2024-05-20:
hab was gefunden:
in history.html, Funktion calcScale(), wird bei der Berechnung von ts_pad anscheinend davon ausgegangen, dass man den oberen Plot an seiner Dauer < 1800 sec erkennen kann:
                s.ts_pad = (s.ts_dur < 1800) ? 0 : s.ts_start % 1800
 Aber wenn man als Abfrage-Intervall z.B. 15 sec eingestellt hat, ist ts_dur = refresh * value.length = 15 * 256 = 3840 > 1800..
 Ich gucke mal weiter..

strepwo, 2024-05-20:
hab vermutlich eine Lösung, siehe patch:
[history_patch.txt](https://github.com/lumapu/ahoy/files/15375412/history_patch.txt)
bisher: der ts_pad wurde so eingestellt, dass der Plot der Messwerte mit dem Plot des Grid zusammenpasst
neu: der Plot des Grid wird mit der korrekten Zeit gemacht; dadurch kann ts_pad entfallen, und dann gibte auch keine Probleme mehr damit. Kurzer Test sieht gut aus: alle Werte da, Zeiten stimmen. Bitte testen

Gubi2023, 2024-05-20:
ich hoffe, das kann jemand übernehmen, ich kanns nicht testen....

lumapu, 2024-05-20:
ich teste und werde es übernehmen, wenn es passt. @strepwo echt cool, dass du dich da reingefuchst hast und den Code 'verstanden' hast. Mir gefällt, dass es jetzt weniger Code als davor ist - wäre ja cool, wenn damit das Problem gelöst wurde.

lumapu, 2024-05-20:
mit 5s Intervall geht's weiterhin, daher empfinde ich es mal als gut und schicke es später ins Rennen, vielen Dank nochmal @strepwo 

[image]

strepwo, 2024-05-20:
ok super 
'Code verstanden' - so halb, ist schon etwas kompliziert.. eher ein 'educated guess' :-)

Was man jetzt noch machen könnte/sollte: die Werte für den unteren Tages-Plot sind momentan einfach jeder x.te Messwert - besser wäre, diese zu mitteln (im C-code), das gibt die Gesamtleistung besser wieder. Oder anders gesagt: ein anti-aliasing einbauen.

lumapu, 2024-05-20:
evtl. willst du diesen Wunsch einfach in #1199 formulieren, dann vergessen wir es auch nicht 😊

Loetnase, 2024-05-20:
Wäre es möglich diese Info, wie auf der Seite Daten, auch mit darzustellen?
[image]
Ich schreibe es auch mal in die #1199

Gubi2023, 2024-05-21:
mit 15 sec- Abfrage funktioniert es jetzt tadellos, gelegentlich teste ich noch andere Abfrageintervalls.

Gubi2023, 2024-05-21:
Test mit 30 und 25 sec. Abfrage ebenfalls ok. 
Viuelen Dank für den schnellen Fix!!