lumapu/ahoy #1636: getNtpTime failed

Hardy3eich, 2024-05-22:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0_.6_.__9
Github Hash: _____15ec6a0__

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hardy3eich, 2024-05-22:
My mistake, I had forbidden the ahoy-DTU from “calling home” in the router a few weeks ago. I took out the lock and now the time is correct again.

Loetnase, 2024-05-23:
Maybe you can take your router as your timeserver. Most of the router can do that.

Hardy3eich, 2024-05-23:
I tried that and though my fritz.box should work as a time server it does not work with these settings

[image]

Loetnase, 2024-05-24:
set your port to 123
[image]
and make the hook in
[image]
i have it working

DanielR92, 2024-06-01:
I think it is now working, I closed this topic. 
Reopen when not pls.

closed 2024-06-01 by DanielR92