lumapu/ahoy #1637: Meine AhoyDTU bootet neuerdings immer wieder neu

RufusRed80, 2024-05-23:
## Hardware
Wemos D1 Mini

### nRF24L01+ Module
nRF24L01+

### Antenna:
external antenna (SMA)

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
AhoyDTU Webinstaller

### Debugging:
USB Serial Log 

`-------------------
Welcome to AHOY!

point your browser to http://192.168.58.14 (Station)
to configure your device
-------------------

I: [NTP]: 2024-05-23 06:03:19 UTC

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020ea93 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffd0 offset: 0150
3ffffee0:  3fff0b78 3fff0f80 4023b108 4023b050  
3ffffef0:  00000001 00000001 3fff0b78 4020f0c1  
3fffff00:  00000005 00000005 3fffff20 3fff3a14  
3fffff10:  0000075b 40223e40 3fff3218 402235cc  
3fffff20:  00000000 00000000 3ffe85ec 3fff0bec  
3fffff30:  3fff0e7c 3fff0f4c 40237eba 3fff0f4c  
3fffff40:  3fff0e7c 3fff0c0c 3fff0b7c 4020595b  
3fffff50:  01010101 01010101 01010101 00000000  
3fffff60:  00000000 00000000 00000000 00000000  
3fffff70:  3fff0f50 3fffff54 40237eba 3fff3a14  
3fffff80:  00000000 00000000 3fff2ab4 3fff3a14  
3fffff90:  3fffdad0 00000000 3fff0b78 4021f328  
3fffffa0:  3fffdad0 00000000 3fff39e8 3fff3a14  
3fffffb0:  00000000 00000000 3fff39e8 40238048  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100551  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00091440
~ld
��n�r��n|�l�l`bbrl�nb�nl`�rl�l�� I: Initializing FS ..
I:  .. done
I: Config Version: 11
I: Settings valid: true
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xe7e7e7e7e7
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0x5712708001
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: DTU_SN: 80701257

-------------------
Welcome to AHOY!

point your browser to http://192.168.4.1 (Password: esp_8266)
to configure your device
-------------------

I: added inverter HM116183063458
E: inverter type can't be detected!
scanning APs with SSID Eselfe
BSSID 0: 7a 95 cd ed c0 c6
connect to network 'Eselfe
try to connect to BSSID: 7a 95 cd ed c0 c6
scanning APs with SSID Eselfe
BSSID 0: 7a 95 cd ed c0 c6
connect to network 'Eselfe
try to connect to BSSID: 7a 95 cd ed c0 c6
I: Network connected

-------------------
Welcome to AHOY!

point your browser to http://192.168.58.14 (Station)
to configure your device
-------------------

I: [NTP]: 2024-05-23 06:03:39 UTC

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020ea93 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd90 end: 3fffffd0 offset: 0150
3ffffee0:  3fff0b78 3fff0f80 4023b108 4023b050  
3ffffef0:  00000001 00000001 3fff0b78 4020f0c1  
3fffff00:  00000000 00000000 00000001 3fff3a14  
3fffff10:  0000075b 40223e40 3fff3218 402235cc  
3fffff20:  00000000 00000000 3ffe85ec 3fff0bec  
3fffff30:  3fff0e7c 3fff0f4c 40237eba 3fff0f4c  
3fffff40:  3fff0e7c 3fff0c0c 3fff0b7c 4020595b  
3fffff50:  01010101 01010101 01010101 00000000  
3fffff60:  00000000 00000000 00000000 00000000  
3fffff70:  3fff0f50 3fffff54 40237eba 3fff3a14  
3fffff80:  00000000 00000000 3fff2ab4 3fff3a14  
3fffff90:  3fffdad0 00000000 3fff0b78 4021f328  
3fffffa0:  3fffdad0 00000000 3fff39e8 3fff3a14  
3fffffb0:  00000000 00000000 3fff39e8 40238048  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100551  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00091440
~ld
��n�r��n|�l�l`bbrl�nB�nl`�rl�l�� I: Initializing FS ..
I:  .. done
I: Config Version: 11
I: Settings valid: true
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xe7e7e7e7e7
pipe 0 ( open ) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0x5712708001
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: DTU_SN: 80701257
`

Ich habe meinen Hoymiles HM-1500 eingetragen gehabt was auch gut funktionierte. Gestern habe ich zusätzlich einen  Hoymiles HMS-800-2T eingetragen (hier liegt glaub ich der Fehler?) und seitdem startet die AhoyDTU immer wieder neu. Was kann ich tun? Der Reboot geht so schnell von statten, dass ich mich nicht aufs Webinterface einloggen kann und den WR wieder löschen kann :(

Was kann ich tun? Ich habe leider keine Sicherung meiner Konfig, so dass ich alles neu eintragen müsste wenn ich neuinstalliere. Wäre schön, wenns ohne Neuinstallation wieder läuft. 

LG 
Markus

Gubi2023, 2024-05-23:
wenn ich deine Konfig richtig interpretiere, hat deine DTU nur ein NRF24-Modul. Für einen HMS-800-2T  brauchst du zusätzlich einen CMT2300A um die Daten zu empfangen. D.h. du musst deine DTU umbauen oder gleich ne andere kaufen. Ich vermute also, du kommst nicht über ne Neuinstallation rum. Wenn du die alte Konfig retten willst, einfach den HMS abgesteckt lassen, vielleicht geht es dann..

RufusRed80, 2024-05-23:
> Wenn du die alte Konfig retten willst, einfach den HMS abgesteckt lassen, vielleicht geht es dann..

Ja das CMT2300A fehlt mir in der Tat wirklich noch, besorge ich :-). Habe die DTU gerade mit ins Büro genommen, also weit weg von den Wechselrichtern, einen Hotspot mit gleichen WLAN-Daten aufgemacht. Auch hier bleibt diese Bootschleife bestehen :(

Gubi2023, 2024-05-23:
ok, vielleicht hast du Glück und die DTU geht wieder, wenn das CMT-Modul angeschlossen ist.

closed 2024-08-16 by lumapu