lumapu/ahoy #1640: Graph on the "small" display shows no values

wurschtbrot01, 2024-05-26:
Hello,
I have the problem that I can activate the graph on my small display and see it, but it remains empty without any "values"

I've already restarted it and reinstalled it
but the same problem always remains
[image]
[image]
[image]

## Hardware
  * [ ] ESP32 ESP32-D0WD-V3

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [yes ] external antenna (SMA)

### Power Stabilization:
* [yes ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Version / Git SHA: 
Version: 0.8.83
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

Dirk-Ahoy, 2024-05-26:
Please enter your coordinates under Sunset in the settings Menu ;)

wurschtbrot01, 2024-05-26:
> Nach dem Nick ja in Deutsch :) Du musst die Koordinaten von dir zu Hause im Menu eintragen sonst geht der Graph am Display nicht.

Hey Hey,
ja Deutsch ;)
alles klar... hab mal Längen und Breitengrade  eingefügt... nur leider sind meine WR schon offline ;)
werde morgen mal berichten.
Vielen Dank schonmal ;)

Loetnase, 2024-05-26:
Installiere dir doch gleich die aktuelle DEV 8.122, da wurde einiges verbessert und verschönert. Auch die online Leistungsanzeige.

wurschtbrot01, 2024-05-27:
Danke an euch!!!
funktioniert jetzt und kann geschlossen werden ;)
[image]

closed 2024-05-27 by wurschtbrot01