lumapu/ahoy #1641: [Bug] YieldDay und YieldTotal werden nicht mehr per MQTT übertragen, wenn ein WR offline ist

wlconsult, 2024-05-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.121

### Github Hash

CC BY-NC-SA 4.0

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Nothing

### Error description

Die Werte für YieldDay und YieldTotal werden nicht mehr per MQTT übertragen, wenn einer der konfigurierten Wechselrichter nicht verbunden ist.  Auf der AhoyDTU-Website werden die Werte aber korrekt angezeigt.

Problem war offenbar schon in 0.6.9 vorhanden (das war mein alter SW-Stand). Ich habe dann auf 0.8.121 geupdated, aber das hat keine Verbesserung gebracht. So bald der nicht verfügbare Wechselrichter wieder mit der AhoyDTU verbunden ist, werden die Werte wieder übertragen.

lumapu, 2024-05-30:
das ist absichtlich so. Die Totalwerte werden nur übertragen, wenn alle Wechselrichter verfügbar sind - sonst kann es uU. Probleme tagsüber geben, wenn mal eine Wechselrichter nicht erreichbar ist. Abends sollten die Werte sowieso nicht weit abweichen, sodass das aus meiner Sicht kein Problem ist.

@knickohr korrigiere mich, falls ich falsch argumentiert habe.

wlconsult, 2024-05-31:
Vielen Dank für die Klarstellung, war mir so nicht bewusst. Evtl. kann man das aber auf die Feature-List setzen, die Website der AhoyDTU zeigt die Daten in so einem Fall ja auch korrekt an.

closed 2024-05-31 by wlconsult