lumapu/ahoy #1642: [Bug]

oremountains, 2024-05-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

08.83

### Github Hash

08.83

### Build & Flash Method

ESP Tools (flash)

### Setup

NRF24L01 | connected

250 kbps

Interval [s]  15
Inverter Gap [ms]  500
Reset values and YieldDay at midnight  x
Reset values when inverter polling pauses at sunset  x
Read Grid Profile x
Yield Efficiency (Standard 1.0)  1

Serial Console
print inverter data  x
Serial Debug  x
Privacy Mode   x

At least one inverter is producing  off
MqTT connected  off
LED polarity  low active

### Debug Serial Log output

```shell
Only version v0.7.36 works for me, the higher ones don't get a connection, it's always sent on different channels, but nothing happens.
  In version v0.7.36 it is still possible to set the power level of the radio module, and works only with  MIN Power level .
```

### Error description

no connection to the inverter

13:35:05.339 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3, s: 0, f: 0, n: 3 | p: 255
13:35:05.342 I: (#0) TX 27 CH75 | 15 01 80
13:35:06.844 I: (#0) RX  40ms | 27 CH23 | 95 81
13:35:06.845 I: (#0) Payload (14)
13:35:06.847 -----
13:35:06.847 I: com loop duration: 508ms
13:35:06.847 -----
13:35:10.339 I: Iv: 0
13:35:10.340 I: 
13:35:15.339 I: Iv: 0
13:35:15.340 I: 
13:35:20.339 I: Iv: 0
13:35:20.340 I: 
13:35:20.341 I: (#0) Radio infos: -6 -6 -6 -6 -4 | t: 4, s: 1, f: 0, n: 3 | p: 255
13:35:20.343 I: (#0) TX 27 CH75 | 15 00 80
13:35:21.845 I: (#0) request timeout: 501ms
13:35:21.846 -----
13:35:21.846 I: com loop duration: 505ms
13:35:21.847 -----
13:35:25.339 I: Iv: 0
13:35:25.340 I: 
13:35:30.339 I: Iv: 0
13:35:30.340 I: 
13:35:35.339 I: Iv: 0
13:35:35.340 I: 
13:35:35.341 I: (#0) Radio infos: -6 -6 -6 -6 -5 | t: 5, s: 1, f: 0, n: 4 | p: 255
13:35:35.343 I: (#0) TX 27 CH75 | 15 00 80
13:35:36.845 I: (#0) request timeout: 501ms
13:35:36.846 -----
13:35:36.846 I: com loop duration: 505ms
13:35:36.847 -----
13:35:40.339 I: Iv: 0
13:35:40.340 I: 
13:35:45.339 I: Iv: 0
13:35:45.340 I: 
13:35:50.339 I: Iv: 0
13:35:50.340 I: 
13:35:50.341 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 6, s: 1, f: 0, n: 5 | p: 255
13:35:50.343 I: (#0) TX 27 CH3 | 15 00 80
13:35:51.845 I: (#0) request timeout: 501ms
13:35:51.846 -----
13:35:51.846 I: com loop duration: 505ms
13:35:51.847 -----
13:35:55.339 I: Iv: 0
13:35:55.340 I: 
13:36:00.339 I: Iv: 0
13:36:00.340 I: 
13:36:05.339 I: Iv: 0
13:36:05.340 I: 
13:36:05.341 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 7, s: 1, f: 0, n: 6 | p: 255
13:36:05.343 I: (#0) TX 27 CH23 | 15 00 80
13:36:06.845 I: (#0) request timeout: 500ms
13:36:06.846 -----
13:36:06.846 I: com loop duration: 505ms
13:36:06.847 -----
13:36:10.339 I: Iv: 0
13:36:10.340 I: 
13:36:15.339 I: Iv: 0
13:36:15.340 I: 
13:36:20.339 I: Iv: 0
13:36:20.340 I: 
13:36:20.341 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 8, s: 1, f: 0, n: 7 | p: 255
13:36:20.343 I: (#0) TX 27 CH40 | 15 00 80
13:36:21.845 I: (#0) request timeout: 501ms
13:36:21.846 -----
13:36:21.846 I: com loop duration: 506ms
13:36:21.847 -----

tictrick, 2024-05-27:
Please upload a image of your wiring and a image of your System NRF24 state.

tictrick, 2024-05-28:
@MetaChuh 

Based on the lack of responses from the inverter, I am assuming that the NRF24 is not wired correctly for the new firmware. Maybe the interrupt pin is not working correctly.

There was a pin exchange at some point.

closed 2024-08-16 by lumapu