lumapu/ahoy #1643: unable to use static IP [Bug]

technics42, 2024-05-27:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.122.

### Github Hash

4ca0562

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

no mqtt

### Error description

Using 0.8.101 I can use static IP - but when using 0.8.122 I am not able to use static IP
DTU is booting and displaying static IP (e.g. 192.168.0.43), but one cannot connect to it.
USB-log says:

I: Network connected
E: failed to set static IP!
E: failed to set static IP!
...

After some resets/reboots, DTU comes up with completely different IP (e.g. 192.168.0.16) and I am able to connect to this one. In settings my set IP is displayed, but not used.
If I save again and reboot, DTU is not reachable - same error: failed to set static IP
I tried different IP, no success - prior versions worked without problems.

technics42, 2024-05-28:
Version 0.8.101 is working fine - no problems to set IP and use this one. (also after downgrading)
When using 0.8.122 I am not able to use static IP - I am always getting error.
After rebooting some times, the DTU comes up with the first free IP assigned by DHCP.

SebiWolze, 2024-05-30:
Same error with 0.8.122 - ESP32 Wroom here

technics42, 2024-05-31:
Thank you - with version 0.8.123 I am able to use static IP again.

DanielR92, 2024-06-01:
Then we can close this topic here? :)

technics42, 2024-06-09:
I am closing it, although I can't use 0.8.123 due to restarting problems - and 0.8.124 was only built for ESP8266 as minimal version, which I don't want to use...

closed 2024-06-09 by technics42