lumapu/ahoy #1644: reset the Yield Total / Unlock hm600 to 800W

aliander11, 2024-05-28:
Is it possible to delete or reset the "Yield Total" score? If I change the Yield Correction at inputs, it looks wrong.

And: Is it possible to unlock the HM600 to 800W like an HM800? Online, some guy offers an inverter tool that can unlock some HM600 models. It's called the "Flole-Inverter Tool Dongle." Maybe someone can rebuild it.

kiu77, 2024-05-28:
Es gibt eine Antwort ;-)
Sie lautet: nein.
Mit einfachen Mitteln wie dieser DTU geht das nicht, weil mehr als 100% nicht drin sind.
Selbst wenn die Leistungshalbleiter das an Mehrleistung vertragen würden, kommst
Du nicht an den Code des Inverters, um den zu ändern oder zu patchen, bzw.
wäre das so aufwendig, dass Verkauf und Neukauf eine genial günstige Lösung ist.

knickohr, 2024-05-29:
Bezüglich des Flole-Mods gibt es aber noch ein ganz anderes Problen :  Sowohl!Ahoy als auch wahrscheinlich OpenDTU ist das aber nicht mehr kompatibel. Die Original Seriennummer des des Inverters wird ihm immer als eine 600er kennen und die Werte entsprechend anzeigen. Kann gut sein das die Werte dann nicht mehr stimmen.

Und nein, wir können nicht für jeden gepimpten WR die Seriennummer als Ausnahme hinzufügen.

dknubben, 2024-05-29:
> Bezüglich des Flole-Mods gibt es aber noch ein ganz anderes Problen : Sowohl Ahoy als auch wahrscheinlich OpenDTU ist das aber nicht mehr kompatibel. Die Original Seriennummer des des Inverters wird ihn immer als eine 600er kennen und die Werte entsprechend anzeigen. Kann gut sein das die Werte dann nicht mehr stimmen.
> 
> Und nein, wir können nicht für jeden gepimpten WR die Seriennummer als Ausnahme hinzufügen.

Interessant, was bedeutet das genau? Wird die Seriennummer permanent geändert oder wie funktioniert die Modifikation?
Evtl. gibt es ja schon Erfahrungsberichte.

lumapu, 2024-05-30:
am besten direkt in Discord bei flole nachfragen

dknubben, 2024-05-31:
> am besten direkt in Discord bei flole nachfragen
Danke Dir. :+1:

stefan123t, 2024-06-05:
The Flole Mod changes the firmware of the inverter so that it has a new max peakpower.
We are using a table of firmware serial numbers to detect the model you are using and determine the max peakpower from the model number.
There are two options to address this issue:
* flole also modifies the serial number to match with the new model specs.
* we introduce an option where you can set an individual max peakpower, which can default to the one determined from the known serial model table.

It should be assumed that e.g. 
* any 1-on-1 like HM-300, HM-400, HM-500 can only be modified to 500 Wpp,
* any 1-on-2 like HM-600, HM-700 and HM-800 can only be modified to 800 Wpp, 
* any 1-on-4 like HM-1000, HM-1200, HM-1600 can only be modified to 1600 Wpp,
if you do not want to exceed the part specs of the condensators, etc.

aliander11, 2024-06-05:
Is there a DIY method to do this yourself?

knickohr, 2024-06-05:
„Jedwede Änderungen an der original Firmware von Hoymiles und die daraus resultierenden Fehler bzw. seltsame Verhaltensweisen werden von Ahoy nicht supportet. Bei Problemen wenden Sie sich bitte an den Entwickler der diese Änderungen an dem Wechselrichter vorgenommen hat.“

UVILG, 2024-06-11:
"Is it possible to delete or reset the "Yield Total" score? " 
Can the inverter total yield score be set to zero ?

lumapu, 2024-06-12:
not with Ahoy - we only can add an offset to the yield that it feels like zero. This can be done in the inverter settings and needs to be set for each input. The inverter itself remains in its original state.

closed 2024-08-16 by lumapu

seba1337, 2025-01-10:
> @aliander11 I believe @Flole998 put considerable effort in the solution of his Flole Tool / Mod (including several fried or bricked inverters) which he has not open sourced yet.
> 
> Maybe you can make a generous donation to cover his expenses and he will contribute the code ?

Where on Discord? BTW is this also for HMS or only HM series?