lumapu/ahoy #1649: HM-1600-4T, no item for limit the power

DV-STUDIOS-DE, 2024-05-30:
Hello,
Version is 240217_ahoy_0.8.83_5ebfe5a_esp32-wroom32.bin
Inverter is a HMS-1600-4T.

In Openhab i get no items for limit the power of the inverter
[image]

HW:
ESP32
DTU-MAX-HMS
CMT2300A
external antenna (SMA)
AhoyDTU Webinstaller

lumapu, 2024-05-30:
can you check our lasted development build? https://fw.ahoydtu.de/fw/dev/

DV-STUDIOS-DE, 2024-05-31:
I have tested DevV 1.22, same result.
In V 1.23 the AP password esp_8266 does not work, so I cant test it

DV-STUDIOS-DE, 2024-06-01:
V 1.23 also tested, same result and new cyrpted signs in MQTT names
[image]

DV-STUDIOS-DE, 2024-06-03:
Lukas, what Do you think to solve this problem?

stefan123t, 2024-11-03:
@DV-STUDIOS-DE did you notice the fine manual: [User_Manual.md](https://github.com/lumapu/ahoy/blob/main/manual/User_Manual.md#power-limit-relative-non-persistent-) ?