lumapu/ahoy #1650: Exception causing reboot with version 0.8.123 [Bug]

technics42, 2024-05-31:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.123

### Github Hash

8d4ca49

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

no mqtt

### Error description

DTU infrequently throws exception leading to reboot - my guess, it's related to webserver (if you try refreshing a page).

Here is the log:

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: sys
sp: 3fffee20 end: 3fffffb0 offset: 0010
3fffee30:  3fffbe74 4022973c 00000144 4023fbbc  
3fffee40:  40277299 3fff2830 3fffbe74 4023021e  
3fffee50:  3fffbe74 3fff2830 3fffba7c 401001e5  
3fffee60:  3fff2830 3fffbe74 00000000 00000000  
3fffee70:  00000000 3fff2830 3fffba7c 402297ea  
3fffee80:  3fff9000 3fff8ff4 3fffb1bc 3fff54d0  
3fffee90:  3fffba7c 3fff54d4 3fff54d5 40255484  
3fffeea0:  40256a80 4028b2b8 00000020 401010c0  
3fffeeb0:  00000010 00000000 00036978 40252225  
3fffeec0:  00000000 00000016 3ffe85f8 3fff5214  
3fffeed0:  3fff5488 3fffb1da 3fffb1bc 4025a861  
3fffeee0:  00000014 3fff5214 00000000 40252225  
3fffeef0:  00000000 000006ef 3ffe85f8 40100d3e  
3fffef00:  402771a8 3fff8ff4 3fff5c5c 3fff7074  
3fffef10:  3fff5214 00000008 3fffb1bc 402518e9  
3fffef20:  3fffdc80 3fff5c5c 3fff8ff4 402516fc  
3fffef30:  4027a36d 3fff5c5c 3fff8ff4 4027a37f  
3fffef40:  3fffb1cc 3fffb1bc 00000000 3fffdab0  
3fffef50:  40276dd7 00000000 3fff8ff4 4027b65f  
[... 251 lines cut ...]
3ffffea0:  40000ea3 00000023 3fffc228 40106455  
3ffffeb0:  4000050c 00000000 3fffd9d0 3fff4e04  
3ffffec0:  4023ffcb 00000030 00000010 ffffffff  
3ffffed0:  4023ffcb feefeffe feefeffe feefeffe  
3ffffee0:  00000000 00000000 00000000 fffffffe  

3ffffef0:  4023d7f4 3fffc6fc 00000001 3fff4dd8  
3fffff00:  00000000 3fffdad0 3fff4e04 00000030  
3fffff10:  00000000 3fffdad0 3fff4e04 00000030  
3fffff20:  00000000 00036921 25e353f7 03f18e53  
3fffff30:  3fff1394 3fff1464 4023fe32 3fff4e04  
3fffff40:  3fffdad0 00000000 3fff1094 4020a4b1  
3fffff50:  01010101 01010101 00010101 00000000  
3fffff60:  00000000 00000000 00000000 3fff0000  
3fffff70:  3fffdad0 00000000 3fff1090 40100169  
3fffff80:  00000000 3fff2090 4020a96c 3fff4e04  
3fffff90:  00000000 00000000 3ffe85f4 00000000  
3fffffa0:  3fffdad0 3fff4e04 4023fe32 3fffefb0  
<<<stack<<<

last failed alloc call: 4023021E(324)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4023021e

lumapu, 2024-05-31:
the exception says that you're out of memory. How many inverters do have configured?

lumapu, 2024-05-31:
It would be of interest which value you see for free heap in `/system`

technics42, 2024-05-31:
2 inverters polling with 15 secs.
Heap after startup is approx. 11000
Refreshing of any webpage (e.g. also system) is triggering exception. (after DTU seems not to be reachable any more)
Tried with different OS and browsers.

I guess it's the webserver-library update...

lumapu, 2024-05-31:
I think it is related to the change in `RestApi.h` there was a change of buffer size. I reverted it in the version below. Can you try this version and give feedback?

[ahoy_esp8266_0.8.123_restapifix.zip](https://github.com/user-attachments/files/15514994/ahoy_esp8266_0.8.123_restapifix.zip)

technics42, 2024-05-31:
I tried this fixed version and got no exception so far.
Webserver seems up and running and responding well - but this is just a "minimal" version. (free heap approx. 17000)
Display and History is not working - so I cannot judge if this really solved the issue...

Zordandi, 2024-06-04:
I tried 240530_ahoy_0.8.123_8d4ca49_esp32-wroom32-de.bin and had similar problems with the webserver. Webpages were loading and reacting very slowly and delayed. Then I went back to 240523_ahoy_0.8.122_4ca0562_esp32-wroom32-de.bin and the webserver works like a charme again.

SvenLuebke, 2024-07-16:
After finding out how to capture the serial output of the ESP8266 when connected to a FritzBox I was now also able to record the C++ exception running the [GIT SHA: 3c0c8ee :: 0.8.129](https://github.com/lumapu/ahoy/commits/3c0c8ee)

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: sys
sp: 3fffe670 end: 3fffffb0 offset: 0010
3fffe680:  3fff2908 3fffb174 000000b0 4024032c  
3fffe690:  3fff2908 3fffb174 3fff89b4 402216ca  
3fffe6a0:  00000000 000c000f 00000000 00000000  
3fffe6b0:  00000000 00000000 00000000 00000000  
3fffe6c0:  00000000 00000000 00000000 00000000  
3fffe6d0:  00000000 00000000 00000000 00000000  
3fffe6e0:  00000000 00000000 00000000 00000000  
3fffe6f0:  00000000 00000000 00000000 00000000  
3fffe700:  00000000 00000000 00000000 00000000  
3fffe710:  00000000 00000000 00000000 00000000  
3fffe720:  00000001 ffffffff 0452f37d 00000001  
3fffe730:  752a0c00 fffffffe 0452f37d 00000001  
3fffe740:  8ad5f400 00000001 0452f37d 00000001  
3fffe750:  00000000 00000000 00000000 00000000  
3fffe760:  00000000 00000000 00000000 00000000  
3fffe770:  00000001 00000000 00000010 00000000  
3fffe780:  00000001 00000000 00000010 00000000  
3fffe790:  0000ba2d 0000d983 00000000 00000001  
3fffe7a0:  00000000 80000000 00000000 0000001b  
3fffe7b0:  000005e0 00000000 401049c1 3ffefb60  
3fffe7c0:  0000003c 00000000 3fffe930 000000fc  
3fffe7d0:  000005e0 00000000 401049c1 3ffefb60  
3fffe7e0:  00000000 00000000 00000020 401005f8  
3fffe7f0:  0000001c 00000001 00000000 40102924  
3fffe800:  3ffeb9e0 40105ba3 3ffef290 401005f8  
3fffe810:  00000001 40104d7e 3ffefb60 401005f8  
3fffe820:  401051ef bc41bb69 3ffefd40 4010368c  
3fffe830:  3ffec26c 00000000 00000000 4010368c  
3fffe840:  40104cb7 00000043 00000000 00040000  
3fffe850:  00000000 40104ab0 00040000 00040000  
3fffe860:  40103b80 00000000 001c001f 00000000  
3fffe870:  40104ab3 00000000 00000000 00040000  
3fffe880:  00000000 2c9f0300 00000000 3fff0078  
3fffe890:  00000000 3fffc200 00000022 3fffc278  
3fffe8a0:  4000bf92 00000030 00000010 ffffffff  
3fffe8b0:  40238630 3fff75c4 3fff75c7 000000ff  
3fffe8c0:  0000ff00 00ff0000 ff000000 00706374  
3fffe8d0:  4024c49c 3fffe9de 000000ff 3fffe9d0  
3fffe8e0:  3fff6124 00000001 3fff6124 00000030  
3fffe8f0:  0000001b 0000001a 00000100 0000006c  
3fffe900:  40238614 3fffe9e9 000000ff 3fffec14  
3fffe910:  3fff4b58 00000000 3fff6124 00000030  
3fffe920:  401055f1 3ffefb60 3ffef290 00040000  
3fffe930:  40104ab3 00040000 00000000 00040000  
3fffe940:  00000000 40104ab0 00040000 00040000  
3fffe950:  40104ab3 00040000 00000000 00040000  
3fffe960:  00000000 40104ab0 00040000 00040000  
3fffe970:  3ffefb60 40103a17 4000050c 3fffc278  
3fffe980:  40103868 3fffc200 00000022 00000000  
3fffe990:  4024c4a5 2c9f0300 4000050c 3fffc278  
3fffe9a0:  3ffeb3e2 3fffc200 00000022 00000008  
3fffe9b0:  00000001 3fff6124 3fffe9d0 40239499  
3fffe9c0:  00000000 3fff4b58 3fffec14 40236304  
3fffe9d0:  4f484108 54442d59 685f0555 04707474  
3fffe9e0:  7063745f 636f6c05 00006c61 00000000  
3fffe9f0:  00000000 00000000 00000000 00000000  
3fffea00:  00000005 00000000 00000020 401005f8  
3fffea10:  00000000 00000000 00000005 40102924  
3fffea20:  3ffeb9e5 40105bbb 3ffef3d0 00000000  
[... 326 lines cut ...]
3ffffe90:  40000f83 00000030 0000001e ffffffff  
3ffffea0:  40000ea3 00000023 00000001 00000001  
3ffffeb0:  3fffdab0 00000000 3fffd9d0 3fff4e14  
3ffffec0:  00000000 00000000 00000001 00000020  
3ffffed0:  00000000 3fffdad0 3fff4e14 00000030  
3ffffee0:  40231fec 3fffff00 00000000 40232460  
3ffffef0:  00000001 00000000 00000001 4023249e  
3fffff00:  00000008 0035a4c8 0ed91687 3e068751  
3fffff10:  3fffdad0 00000001 3fff5008 3fff4e14  
3fffff20:  3fffdad0 3fff1f40 3fff1d68 4020cbc8  
3fffff30:  000f4240 00000001 3fff1df0 4022809a  
3fffff40:  3fffbad8 3fffbad8 00000000 4020a4b5  
3fffff50:  3fff23f0 3fff396c 3fff72f4 3fff74e8  
3fffff60:  3fff74e8 3fffbacc 00000008 3fff72f4  
3fffff70:  3fff72f4 00000000 3fff1070 40100169  
3fffff80:  00000001 3fff1f40 00000020 3fff4e14  
3fffff90:  00000000 00000000 3ffe85f4 00000000  
3fffffa0:  3fffdad0 3fff4e14 402405a2 3fffefb0  
<<<stack<<<

last failed alloc call: 402216CA(176)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Of course I had these issues with different power supplies, cables and capacitors connected.

stefan123t, 2024-07-19:
@SvenLuebke please enclose your code block in [/] Slash commands Details.

Regarding the Exception you posted you have to Decode it using the [ESP Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) in Arduino. You will also need the binary / header files in order to decode the symbols on the stack.
Also it remains usually quite difficult / impossible to determine the actual cause for such a reboot based on the Stack trace resulting. But thanks for trying ... [I have been there](https://github.com/lumapu/ahoy/issues/15#issuecomment-1115118899), but they didn't sell T-Shirts

SvenLuebke, 2024-07-20:
@stefan123t Do you really mean me? I tried to do it, but the result is exactly the same (starting with three `)

Thanks for the hint. I will try to decode the exception using the decoder.

steve-zero, 2024-07-20:
Die aktuelle 0.8.129 ist bei mir noch immer instabil.
Bei mehrfachem Wechsel der Bereiche (Daten/Verlauf/Einstellungen, etc.) auf dem Webinterface oder Browser-Reloads crasht der 8266 immer wieder.
Besonders der Tab "Verlauf" ist fragil und kämpft mit Ladeverzögerungen, die dann oft zum Crash führen.

Ich bin jetzt auf die 121 zurück. Das ist für mich die letzte stabile (und performante) Version, die ich getestet hatte.
Die 123, 125, 126, 127 , 128 und 129 hängen sich bei mir immer wieder auf, sobald ich das Webinterface nutze.

steve-zero, 2024-07-20:
[Sorry for the German. Now the same feedback in English.]

The current 0.8.129 is also instable for me.
When using the webinterface and reading differen sections of it (Data, History, Settings, etc.) or when reloading with the browser, my 8266 regulary crashes.
Especially the section "History" is very fragile, has loading-delays and often causes crashes.

I am now back to the 121 that I previously tested. This is the last stable and performant version I had.
The 123, 125, 126, 127, 128 and 129 keep crashing when using the webinteface.
The 122 might be stable as suggested, but I haven't tried that one yet.

I am not so familiar with ESP microcontrollers (started with 68hc11 and did Arduinos lately), but I can try to test or check something if I get a hint (what and how).

SvenLuebke, 2024-07-21:
@steve-zero

Thanks for the hint, 'll also try that version!

I'm glad, that I'm not the only one with these problems. But...as some people seem to be very happy, I thought about whether my ESP8266 (from AliExp***) might be corrupt, but I already also changed the complete ESP8266 board (although same AliExpress order) and NRF24L01 board with no improvements. So it might be
* a chip defect (for chips/boards sold at AliExp***)
* other "environmental" issues (local WiFi, connected router etc.)
* issues in this software?
* a combination of all :-)  

I'll try the 121 now and will record the serial again....

BR,
Sven

steve-zero, 2024-07-28:
I can now confirm, that the web-interface is broken since the 123 version.
I tested the 121 sucessfully for a week now with no problems.
It has a MqTT bug however that causes reboots after 1 or 2 days, but when deactivating MqTT, it is rock solid.

So apparently the updated AsyncWebserver library to 3.2.2 does not work properly on a ESP8266.

stefan123t, 2024-07-31:
@SvenLuebke thanks for using the triple backticks (before and after the code block), this formats the code in pre-formatted / ASCII mode. 

If you use the [/] Slash command Detaila it will add some html code to collapse your text / code block to make the issue even more readable.

@steve-zero thanks for hinting to your use of the ESP8266. A lot of the pages served by the ESPAsyncWebServer have to be put together by concatenating large string objects in the ESP memory. This can easily exceed the Heap/Stack for these tiny MCUs. 

So yes this is a Bug and should be treated as such, but I expect that the only apparent solution could be to diaable aome of the functionality on the ESP8266 vs the beefier ESP32.

@lumapu may eg need to restrict the History graph to the ESP32 port/platform. But also the MQTT messagea have to be string'ed together.

technics42, 2024-08-01:
I got V0.8.101 now running for 71 days without reboot - polling 2 inverters  (10s), no mqqt, but requesting values by rest-api (each 10s one inverter-data, alternating).
Not using web-interface very often...
If using it often, it also may reboot...

stefan123t, 2024-11-03:
@technics42 @Zordandi @SvenLuebke @steve-zero could you please try the current v0.8.152 from the development03 branch and report back if you still face the Exception causing reboot, aka Out Of Memory ?

steve-zero, 2024-11-03:
Without MQTT it seems to be stable, same as v0.8.121. No long term experience though.
With MQTT, after three to maximum four clicks on the webinterface, it crashes. Once crashed twice within a few seconds, it lost all configuration and i had to power cycle it. After the first crash it was asking for a Wifi-Setup while already connected via Wifi.
Interestingly, with MQTT on, the web-console is red and stays red. After trying to open the console, there are delays in loading the webinterface. The memory consumption looks fine, at least the heap is around 11 to 12kB. it crashes after the next click anyway, but even when avoiding the console. Then it is four clicks to crash.
Without MQTT the console is instantly green.
I know, the above is not scientific, but a rough tryout within the 10 minutes a have. No improvement over the .121 but maybe better than the .140 as it could at least run stable without MQTT. Can tell more about that in 2 days.

stefan123t, 2024-11-04:
@steve-zero how many inverters and which polling interval do you have in your configuration ?

technics42, 2024-11-04:
I already used 0.8.152 some days ago, but encountered reboots/crashes and freezing webserver. See my reported issues.
Switched back to 0.8.101, which is working well for me.

steve-zero, 2024-11-05:
> @steve-zero how many inverters and which polling interval do you have in your configuration ?

One HM-1500, polling every 15 seconds, as per default.

Ohne MQTT lief er übrigens seit gestern problemlos und stabil durch. Ich finde es gut, dass die 152 immerhin so wieder stabil läuft, aber MQTT zusammen mit dem Verlauf und/oder der Konsole ist wohl zu viel für den kleinen 8266...

SvenLuebke, 2024-11-12:
I'm also still facing reboot issues: While on "Live" page clicking on main page (to see the uptime) doesn't work after some days, although "Live" was updating without any issue. The ESP reboots and main page is working afterwards.