lumapu/ahoy #1651: V0.8.123: nach 6 std. nicht mehr erreichbar[Bug]

Gubi2023, 2024-05-31:
### Platform

ESP32

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.123

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

updated um 2 uhr von 0.8.122

### Error description

nachdem die 0.8.122 ohne Probleme 6 Tage durchlief, ist die 0.8.123 schon nach 6 St. nicht mehr erreichbar, auch per ping nicht, obwohl sie in der Fritzbox noch angezeigt wird; auch ein WLAN aus/ein der Fritzbox, das die Vorgängerversion wieder erreichbar machte, blieb erfolglos..

Gubi2023, 2024-05-31:
[image]
[image]

technics42, 2024-05-31:
I guess it's related to the exceptions I am encountering. Maybe if you try to reach DTU by browser, it may cause a reboot and DTU would be available again (for some time).

Dirk-Ahoy, 2024-05-31:
Also meine 4 DTUs sind alle durchgelaufen seit heut morgen 6:00 - alle erreichbar und kein reboot.

dtuuser, 2024-05-31:
Meine auch seit dem Update vor 14 Stunden stabil ohne Reboot

Tolbag, 2024-06-12:
Mein ESP8266 ist auch alle paar Tage, oft gegen Nachmittag, nicht mehr erreichbar. Netzteil an/aus startet ihn komplett neu, dann geht es sofort wieder.
Neugestartet per Software wird immer um Mitternacht.

lumapu, 2024-06-12:
wie gut ist bei euch der WLAN Empfang?

Gubi2023, 2024-06-12:
sehr unterschiedlicher Empfang, zwischen 1-3 Balken..
Hab jetzt die .124 drauf, gestern 3 Reboots, heute lief sie durch.
könnte es damit zusammenhängen, ob man die Webseite aufruft? Diesen Eindruck hatte ich manchmal.

Gubi2023, 2024-06-13:
heute morgen 7:47 Uhr startete sich DTU wieder neu. Hab mal in HA den entsprechenden Zeitraum angeschaut, keine Auffälligkeiten. k.A. warum sie rebootet hat...

Gubi2023, 2024-06-21:
tja, bin grad im Urlaub, also keiner zuhause, aber meine DTU (V.126) resettted sich trotzdem fröhlich  3-4mal am Tag vor sich hin. Ärgerlich, da dann ständig mein Verlauf beim Teufel ist.

cibis8, 2024-06-21:
Bei mir auch sehr viele Reboots, bin wieder zurück zur .110, da danach ja das DHCP nicht mehr ging. aktuell bleibe ich bei der 110. die lief ohne Probleme auch mehrere Tage/Wochen.

Dirk-Ahoy, 2024-06-21:
[image]

Kann mich nicht beschweren - das ist noch 123 -die 126 läuft auf 2 Fusion Board auch ohne Reboot bis jetzt.

Gubi2023, 2024-06-29:
Ich habe jetzt beobachtet, dass ide Empfangsstärke zwischen DTU und WR kontinuierlich abnimmt. Anscheinend startet sich die DTU neu, wenn kein Empfang mehr da ist, die Empfangsstärke ist dann wieder auf max.  
[image]
[image]

Gubi2023, 2024-06-29:
> wie gut ist bei euch der WLAN Empfang?
Ich habe jetzt beobachtet, dass die Empfangsstärke zwischen DTU und WR kontinuierlich abnimmt. Anscheinend startet sich die DTU neu, wenn kein Empfang mehr da ist, die Empfangsstärke ist dann wieder auf max.
Screenshot 2024-06-29 161622
[image]

lumapu, 2024-07-07:
meinst du wirklich die Empfangsstärke der Wechselrichter oder evtl. den des WLAN Moduls? Ich habe beobachtet, dass bei schlechter WLAN Verbindung der Speicher durch MqTT von Ahoy vollläuft und den Reboot verursacht.

Dirk-Ahoy, 2024-07-08:
Bei 1Mbits vom Repeater zur Ahoy kein Wunder -  solltest du mal was näher an die Fritzbox bringen oder an den Repeater.

Gubi2023, 2024-07-08:
Das Teil steht direkt neben einer Fritzbox7560,  k.A. warum es sich nicht mit der verbindet....

Gubi2023, 2024-07-08:
Das Ding wir über mesh immer wieder mit nem anderen Repeater verbunden....
bin inzwischen auf 127

lumapu, 2024-08-13:
@knickohr, du hattest irgendwas ähnliches geschrieben, dass AVM ein neues Feature eingeführt hat was evtl. zu diesem Verhalten führt.
Ich denke es war der hier:

* https://github.com/lumapu/ahoy/issues/1596

knickohr, 2024-08-14:
Ja, genau das ist es !

Nochmal zur Verifikation :  Welche FW läuft auf der Fritte ?

Gubi2023, 2024-08-14:
ich habe immer noch 2-3 reboots täglich, aber wie es aussieht, liegt es daran: MqTT läuft voll und bekommt die Pakete nicht abgeschickt, Ahoy bricht dann zusammen, da kein Speicher mehr da ist (siehe #1686) Deshalb mach ich den Issue zu und hoffe auf Beseitigung des MqTT-Fehlers.

closed 2024-08-14 by Gubi2023

knickohr, 2024-08-14:
Das Problem MQTT sollte ab der 124, spätestens ab der 126 behoben sein. Es reconnected zwar weiterhin den Broker, aber es sollte nicht mehr booten. Außer der Broker meldet sich nicht mehr.

Gubi2023, 2024-08-14:
ich habe Lukas am 7.7. nen coredump mit .127 geschickt, da war der Fehler noch drin...