lumapu/ahoy #1655: [Bug] HMS 400 wird mit 2 Channels angezeigt

tester277, 2024-06-02:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.123

### Github Hash

-

### Build & Flash Method

was already installed

### Setup

OpenDTU Fusion mit bisher 4 HMS Geräten (eins davon ebenso mit einem Channel - 
[image]
[image]
HMS500)

### Error description

Bei meinem neu eingerichteten HMS werden 2 Channel angezeigt, obwohl diese Geräte nur über einen Channel verfügen.
Bei einem (nicht mehr vorhandenen) HMS 400 wurde es in der Vergangenheit korrekt mit einem Channel angezeigt. Könnte es an der relativ neuen HW Version liegen?
Was kann ich noch liefern?
Ist für mich kein akutes Problem.
Danke für eure Mühe!

Gubi2023, 2024-06-02:
hast du vvlcht unter Einstellungen-WR-Bearbeiten ein 2.Modul eingetragen?

tester277, 2024-06-02:
Nein da stimmt die Ansicht. Es lässt nur eine Eingabe zu.
[image]

tester277, 2024-06-02:
Aus meiner Sicht scheint die Seriennummer das „Problem“ zu sein.
Im JSON Settingsfile werden initial alle Werte so gesetzt: "pwr":[300,400,400,400,400,400],
Der Workaround wäre ja klar.

lumapu, 2024-06-09:
Die Seriennummer beginnt mit `1124`, richtig? Nach der Theorie von @tester277 müssten doch 6 Kanäle angezeigt werden?. Über diese Option kann man lediglich Inputs ausblenden, die nicht verwendet werden.
Ich denke eher, dass irgendwo die Anzahl der Kanäle falsch interpretiert wird.

Leider gab es auch schon einen HMS-400 mit der Payload von 2 Kanälen, aber tatsächlich nur einem Eingang, die Seriennummer beginnt bei diesem mit `1125`, siehe 
* https://github.com/lumapu/ahoy/issues/1460

tester277, 2024-06-09:
Die Seriennummer vom HMS 400 (mit dem nur Channel bei Erstkonfiguration angezeigt wurde) begann mit 1124 - richtig.

lumapu, 2024-06-09:
ich bin verwirrt. Geht es um einen HMS-400 oder HMS-500 der falsches anzeigt?
Es geht um die Seriennumer oben mit `1125xxx` oder?

tester277, 2024-06-10:
@lumapu es geht aktuell um einen HMS 400 mit SN beginnend mit 1125. zum Vergleich hatte ich einen HMS 400 genannt, der eine Serienummer beginnend mit 1124 hat und bei Einrichtung mit einem Channel angezeigt wurde. Ein HMS 500 den ich bisitze,hat ebenfalls eine SN beginnend mit 1124.

lumapu, 2024-08-13:
ist dieser Fehler noch präsent?

tester277, 2024-08-15:
Mit [0.8.124] getestet: der besagte HMS 400 mit SN beginnend mit 1125 wird mit 1 Channel hinzugefügt, wenn ich ihn neu anlege. So soll es sein! Danke schön

closed 2024-08-15 by tester277